MCUdude/MegaCoreX #3: Testing with Xplained Pro 4809

MCUdude, 2019-03-04:
I do own a Curiosity Nano and an AVR-IOT WG, but not an Xplained Pro 4809. It would be great if someone with this board could verity that they were able to upload and print to the serial monitor using this board.
@westfw do you own an Xplained Pro 4809? If so, would you like to give it a quick test with this core?

AndersDenmark, 2019-03-23:
Hi I do have 2 ATMEGA 4809 Xplained pro. and I am sorry, but i was not able to get the serial print to work on this bord, tried both of them. I even tried to go through all serial ports with no luck ( serial.print, serial1.print, serial2.print, serial3.print ). But i can see the orange led goes on, as soon as i start the serial monitor in the Arduino IDE. tryed different BAUD rates of curse both for the uploaded program, and the serial monitor. "always chose same BAUD rate both places". tried all different clocks. is it not possible to make the programmer to choose the external clock crystal 32.76kHz, i think it is the same for Arduino Uno wifi rev2. i do also own one board of those, will compare bords later on. i have tested same configuration on the ATMEGA 4809 Curiosity nano, and everything works fantastik, i can serial print and program the chip, blink the onboard LED_BUILTIN Arduino pin 39, all seems ok so far need to do some more testing with hardware on the pins. also here it would be nice to be able to use the external crystal on the board. I need help, or some good suggestions on how to program an Arduino und so it will work as one of the three programmer options in the MegacoreX, so it can program the chip on a breakout board, and also use the serial monitor, thinking maybe modifying a nano 4809 to do this with. Idea is to use the 4809 on my own PCB in the future. I am doing this already with the MegaCore, works perfect like this. please let me know if there are anything i can do to be helpful about the mentioned issues.

egilkv, 2019-07-26:
I have two Curiosity Nano boards running under Linux. I could not get the nEDBG thing in Arduino to work, but had no problem uploading a trivial program (blinky plus Serial3) as hex to the nest little file system programmer. Bug files, have not tested more, but can do more, but would appreciate some pointers.

I also have made a small stand-alone m4809 test board myself which I program with a JTAG2UPDI (had originally intended to use the Curiosity Nano mEDBG, but see above). The JTAG2UPDI running on an old Nano worked first time without issues.

KHeintz, 2020-05-07:
Looks to me like serial on the 4809 only works at 20MHz.

MCUdude, 2020-05-07:
That's most likely due to the internal oscillator not being accurate. I have several 4809 boards, and all of them work perfectly at 16 MHz, Serial included.

You could try to tweak the `baud_setting` variable to see if increasing or decreasing it helps.

https://github.com/MCUdude/MegaCoreX/blob/4e712716f00bb66337dee3efd58518fa7c788820/megaavr/cores/coreX-corefiles/UART.cpp#L214-L217

Note that the baud value is 16 bit.

KHeintz, 2020-05-09:
The answer, the 16MHz* clock is 16% too high. The 20MHz clock is exactly right.
I use a quartz oscillator instead, because I also need an exact timer.
*At 5 pieces 4809 DIP40.

MCUdude, 2020-05-09:
16% too fast?! is that even possible? There may be some oscillator calibration procedure you can perform to get it more accurate. No wonder why serial didn't work.

rocketscream, 2020-05-09:
I have tested with 4808 at 8MHz, 1200 bps up to 115200 bps works just fine. Board running at 3.3V, so I can't confirm it will work reliably at 16 MHz.

KHeintz, 2020-05-09:
I have measured the baudrate with a logic analyzer, there comes out ~140000 bit/s at 16 MHz.  ~17% too high.

MCUdude, 2020-05-09:
You can measure the actual clock frequency by setting the CLKOUT bit in the MCLKCTRLA.

```c
MCLKCTRLA |= CLKCTRL_CLKOUT_bm;
```

Not sure if you'll have to use protected write on this or not though. Alternatively:

```c
_PROTECTED_WRITE(CLKCTRL_MCLKCTRLA, CLKCTRL_CLKOUT_bm);
```

KHeintz, 2020-05-09:
I already called this in Setup.
 cli();
_PROTECTED_WRITE(CLKCTRL_MCLKCTRLA, CLKCTRL_CLKOUT_bm);  
sei();
This works.

MCUdude, 2020-05-09:
> This works.

And what do you measure your main clock to be?

KHeintz, 2020-05-09:
Always 20 MHz. I uninstalled MegaCorex once more and then reinstalled it again. Now 16MHz, 8 MHz etc. works. The fuses were not set before. So the 4809 always ran in 20MHz.

MCUdude, 2020-05-09:
> Always 20 MHz. I uninstalled MegaCorex once more and then reinstalled it again. Now 16MHz, 8 MHz etc. works. The fuses were not set before. So the 4809 always ran in 20MHz.

Huh, that's weird. Great to hear it's working now though!

MCUdude, 2020-06-30:
I'm closing this issue because this core has been tested on all supported chips and proven to work. If you're experiencing issues with your hardware and you believe it's MegaCoreX's fault, please open an issue.

closed 2020-06-30 by MCUdude