MCUdude/MegaCoreX #4: Serial on Curiosity Nano

sstaub, 2019-03-05:
I tried various print variants to serial with no luck.
Arduino 1.8.8 on macOS Mojave

Good work:)

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.begin(9600);
  }

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);
  delay(500);
  digitalWrite(LED_BUILTIN, LOW);
  delay(500);
  Serial.println("hello"); // don't work
  printf("hello world\n"); // don't work
  Serial.printf("hello world\n"); // don't work
  }

MCUdude, 2019-03-05:
If you [read the instructions for Curiosity Nano](https://github.com/MCUdude/MegaCoreX#curiosity-nano) you'll see that you will have to use Serial3 as object. 

``` c

Serial3.print("Hello World");
Serial3.print(F("Hello World");
Serial3.printf("Current millis = %d", millis());

```

sstaub, 2019-03-05:
Found it, Thank you.

closed 2019-03-05 by sstaub