MCUdude/MegaCoreX #6: I2C fail and crashe a slave ATMEGA4809 when i2cdetect is launched by the master

goodchip, 2019-03-06:
I've a Raspberry 3 board in I2C master configuration, wired at 3 arduino boards :
 - 2 arduino pro mini (as control sample device) at 0x03 and 0x04 slave i2c address
 - 1 arduino Uno WiFi rev 2 (as 4809 test device) at 0x11 slave i2c address

_NB: the crypto chip on the Uno WiFi Rev 2 has been desolded to remove any potential influence and does not appear in the addresses_

The 3 boards have the same demo i2c sketch in his EEPROM :
```
#include <Wire.h>

void setup() {
  Wire.begin(0x~thisSlaveAddr~);
  Wire.onRequest(reqIRQ);
}

void loop() {
  delay(1000);
}

void reqIRQ() {
  Wire.write(0x01);
}
```

IMPORTANT : I use the Wire.h library I submitted in pull request to make the sketch working  : https://github.com/arduino/ArduinoCore-megaavr/pull/22
---

I use the following command on Pi3 to list the slaves presents on the bus:
> i2cdetect -y 0

When I launch this command for the 1st time, it's happening: (normal condition)
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          03 04 -- -- -- -- -- -- -- -- -- -- -- 
10: -- 11 -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
```

But, at the 2nd launch, the 4809 chip disappear:
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          03 04 -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
```

And, at the 3th launch, the i2c BUS freeze totally for undetermined time:
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
```

A simple reset of the Uno WiFi rev 2 board, restore the freeze.

Interestingly, the following command does NOT cause the problem:
> i2cdetect -r -y 0

The problem only concerns the i2cdetect command, the other master / slave communications seem to work properly.

---
---
---

**Note that the same problem seems to be present even using only Arduino boards:**

I've reproduce the issue only with 1 arduino Nano and my Arduino Uno WiFi Rev2:

```
15:28:41.889 -> Scanning...
15:28:41.889 -> I2C device found at address 0x08  !
15:28:41.922 -> done
15:28:41.922 -> 
15:28:46.878 -> Scanning...
```

**After the 1st scan, the BUS crashes: SDA is continuous at the level 0V and SCL too.**

_NB: the ATMEGA328P emulation is off at the compilation._

---

Sketch on Arduino Nano : (MASTER, I2C scanner)

```
// --------------------------------------
// i2c_scanner
//
// Version 1
//    This program (or code that looks like it)
//    can be found in many places.
//    For example on the Arduino.cc forum.
//    The original author is not know.
// Version 2, Juni 2012, Using Arduino 1.0.1
//     Adapted to be as simple as possible by Arduino.cc user Krodal
// Version 3, Feb 26  2013
//    V3 by louarnold
// Version 4, March 3, 2013, Using Arduino 1.0.3
//    by Arduino.cc user Krodal.
//    Changes by louarnold removed.
//    Scanning addresses changed from 0...127 to 1...119,
//    according to the i2c scanner by Nick Gammon
//    http://www.gammon.com.au/forum/?id=10896
// Version 5, March 28, 2013
//    As version 4, but address scans now to 127.
//    A sensor seems to use address 120.
// Version 6, November 27, 2015.
//    Added waiting for the Leonardo serial communication.
//
//
// This sketch tests the standard 7-bit addresses
// Devices with higher bit address might not be seen properly.
//
 
#include <Wire.h>
 
 
void setup()
{
  Wire.begin();
 
  Serial.begin(9600);
  while (!Serial);             // Leonardo: wait for serial monitor
  Serial.println("\nI2C Scanner");
}
 
 
void loop()
{
  byte error, address;
  int nDevices;
 
  Serial.println("Scanning...");
 
  nDevices = 0;
  for(address = 1; address < 127; address++ )
  {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.
    Wire.beginTransmission(address);
    error = Wire.endTransmission();
 
    if (error == 0)
    {
      Serial.print("I2C device found at address 0x");
      if (address<16)
        Serial.print("0");
      Serial.print(address,HEX);
      Serial.println("  !");
 
      nDevices++;
    }
    else if (error==4)
    {
      Serial.print("Unknown error at address 0x");
      if (address<16)
        Serial.print("0");
      Serial.println(address,HEX);
    }    
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
  else
    Serial.println("done\n");
 
  delay(5000);           // wait 5 seconds for next scan
}
```

Sketch on Arduino Uno WiFi Rev2: (SLAVE, I2C RECEIVER)
```
// Wire Slave Receiver
// by Nicholas Zambetti <http://www.zambetti.com>

// Demonstrates use of the Wire library
// Receives data as an I2C/TWI slave device
// Refer to the "Wire Master Writer" example for use with this

// Created 29 March 2006

// This example code is in the public domain.

#include <Wire.h>

void setup() {
  Wire.begin(8);             // join i2c bus with address #8
  Wire.onReceive(receiveEvent); // register event
  Serial.begin(9600);           // start serial for output
}

void loop() {
  delay(100);
}

// function that executes whenever data is received from master
// this function is registered as an event, see setup()
void receiveEvent(int howMany) {
  while (1 < Wire.available()) { // loop through all but the last
    char c = Wire.read(); // receive byte as a character
    Serial.print(c);         // print the character
  }
  int x = Wire.read();    // receive byte as an integer
  Serial.println(x);         // print the integer
}
```

MCUdude, 2019-03-06:
Thanks for reporting it here too. I'm watching the main repo to see if you guys are able to find out whats causing the issue

goodchip, 2019-03-12:
Issue solved in official MEGAAVR repo.

MCUdude, 2019-03-12:
Awesome! I'm at work at the moment, so I'm not able to handle it just yet. Do you want to submit a PR? 🙂

goodchip, 2019-03-12:
Yep, I PR quickly.

goodchip, 2019-03-12:
Solved with #8

commit 30c3ad25 (2019-03-12): Merge pull request #8 from goodchip/master

Solve issue concern slave reliability during i2c response
Closes #6
closed 2019-03-12 by MCUdude