MCUdude/MegaCoreX #7: Support of RTC Time / RTC interrupts

goodchip, 2019-03-07:
Look at this updated library for use internal RTC clock of Arduino Uno WiFi rev 2 and Microchip curiosity nano board :
https://github.com/goodchip/Time

Example here: https://github.com/goodchip/Time/tree/master/examples/TimeInternalRTC

Features:
 - use internal RTC circuit for real clock in board (not based on millis() function)
 - attach/detach interrupt function to run user function at 1Hz synchronized by RTC
 - 100% compatible with 328p code (run with out RTC) if interrupt are not used

Current limitation:
 - use only external quartz for the moment.
 - alpha version

To do:
 - milisecond syncho RTC function

Have fun!

Sylvain

MCUdude, 2019-03-07:
Awesome! You should submit it as a PR to the main repo when everything is ready. It's great to have megaAVR-0 support in the most common 3rd party libraries.

goodchip, 2019-03-08:
Yes the PR is planned, I still have to test the code ;)

I made an update today that allows a precise start of the second when calling setTime.

If you ever have a moment to test (I added an example sketch about compatibility 328p / 4809).

goodchip, 2019-03-20:
**New features:** 
- adjustable IRQ call frequency added (from 1s to 1/8192s)
- adjustable synchro precision RTC added (from 1s to 1/8192s)
- new maths routines very optimized

@MCUdude: Could you test this version for me? :)

MCUdude, 2019-03-20:
Sure! I'm quite busy at the moment, but I'll test it in a few days.

goodchip, 2019-04-10:
*Major upgrades - stable version*

New! 2 interrupts features:
- InternalRTC.attachInterrupt: attach a IRQ user function called at the freq frequency
- InternalRTC.attachClockInterrupt: attach a IRQ user function called each new second

Upgrades:
- removal of the function setPrecision(), now useless
- now always maximal RTC cycles precision without cpu load

@MCUdude: This version is very *hot* to test it :D

MCUdude, 2019-04-10:
Cool! I'll _promise_ I test it tonight. Right now I'm on the go.

MCUdude, 2019-04-10:
I just tried both internalRTC examples (had to rename the files into .ino files and add folders for each sketch. None of them seemed to work.

TimeInternalRTC_compatibility just printed the time initialized in setup().
TimeInternalRTC_interrupt didn't do anything at all.

I'm using an Arduino UNO Wifi Rev2 for testing.

goodchip, 2019-04-10:
Hum strange things.

Modified versions of timelib.h and time.cpp compile without problem?

MCUdude, 2019-04-10:
I just downloaded your fork as a ZIP file and placed it in the libraries folder

goodchip, 2019-04-10:
Hum.... very strange.

Look if the library is correctly executed.

I look at this tomorow. Sorry.

goodchip, 2019-04-10:
Try with A328p compatibility mode disabled?

MCUdude, 2019-04-10:
> Try with 328p compatibility mode disabled?

Tried to turn compatibility mode off. Still no difference...

goodchip, 2019-04-10:
The RTC registers seem not stars hardware interruptions, strange.

Your Uno board is in right conditions? The quartz is present?

Could you test with a Curiosity Nano? (He works fine here with this board)

goodchip, 2019-04-11:
Hello @MCUdude 

I just installed the fork (just unzipped the archive and put the Time-Master folder in Arduino/Libraries/) on a blank Arduino installation (just installed the official Arduino Uno WiFi Rev2 board).

It works very well on my side (just like with your MegaCoreX).

Could you check again?

PS: maybe trying to reflash the bootloader?

MCUdude, 2019-08-19:
This was solved in the [official megaavr repo](https://github.com/arduino/ArduinoCore-megaavr/issues/27).

closed 2019-08-19 by MCUdude