MCUdude/MegaCoreX #9: Should we re-do the default atmega4809 pinout?

MCUdude, 2019-04-24:
It turns out that Microchip is releasing a [DIP-40 variant](http://ww1.microchip.com/downloads/en/DeviceDoc/40-pin-Data-Sheet-megaAVR-0-series-DS40002104A.pdf) of ATmega4809! 🎉 
This is definitely good news, but the problem is that they had to sacrifice some IOs since the DIP package has 8 fewer pins that the TQFP/QFN variant.

Personally, I was hoping that they tried to make it as close to ATmga3208/4808 as possible, but that doesn't seem to be the case. Some pins here and there is just gone.

The question is: should I/we develop a new "Standard" pinout for the 4809 that matches the DIP variant better? What would this pinout look like? Ideally, I'd like a pinout that "made sense" on both packages.

egilkv, 2019-07-24:
My 5 cents: The current 48 pin pinout seems very logical, so I would not compromise that to support a DIP-40 variant that may not be used much.

AndersDenmark, 2019-07-25:
I too agree I don’t think the 40 pin DIP will be used that much. I am so happy to have a Arduino based IDE tool for the 4809. As I’m using the 4809 chip on most of my personal project PCB,s. I Hope the ability to use external Chrystal soon will be possible to use. I really liking this tool, really nice job.

MCUdude, 2019-07-31:
You guys are right. The DIP-40 chip isn't probably going to be hugely popular. I'm probably going to make a pinout diagram of it, but the pinout is based on the 48-pin variant.

closed 2019-07-31 by MCUdude