MCUdude/MegaCoreX #11: Crashes when using PWM on pin TCB0 and delay() function

goodchip, 2019-06-03:
Hi!

I just located a permanent crash using in same time the PWM feature on pin TCB0 and the delay() function.

The bug is present so much on 4809 and 4808 microcontrollers, but only the pin based on the TCB0 count register is concerned.

*The following sequence crash the µC:*
- set a PWM value to the pin
- waiting a delay for x ms
- set again a PWM value to the pin

I add a sketch to easily reproduce the crash.

NB: the bug was *NOT* present with the genuine Arduino board set for Uno WiFi rev2.

Thanks! :)
Sylvain

```
// bug test for Uno WiFi Rev2 (4809)
//#define testPin 3 // <-- work good
//#define testPin 5 // <-- work good
#define testPin 6 // <-- CRASHES!
//#define testPin 9 // <-- work good
//#define testPin 10 // <-- work good

// bug test for Curiosity Nano (4808)
//#define testPin 10 // <-- work good
//#define testPin 11 // <-- work good
//#define testPin A13 // <-- work good
//#define testPin A14 // <-- CRASHES!

void setup() {

  Serial.begin(9600);
  pinMode(testPin, OUTPUT);

}

void loop() {

  Serial.print("crashed? ");

  analogWrite(testPin, 127);

  Serial.print("hum... ");
  
  delay(10);
  //delayMicroseconds(10000);

  Serial.println("...not this time :)");

}

```

goodchip, 2019-06-03:
After investigation, the bug has been fixed for the Uno WiFi AND 48pin versions (but *not* for the 32pin 4808 version) in the following: commit:https://github.com/MCUdude/MegaCoreX/commit/56474405853f08c9f240e030200ce6bd77f3423b

Set the millis to the timer TIMERB2 (or TIMERB1) resolve the problem on 4808 32pin:
https://github.com/MCUdude/MegaCoreX/blob/56474405853f08c9f240e030200ce6bd77f3423b/avr/variants/32pin-standard/pins_arduino.h#L32

But... set TIMERB0 on Uno WiFi AND 48pin versions of pins_arduino.h do *not*  cause the problem.

MCUdude, 2019-09-10:
Thanks for reporting! I haven't worked on this project for a couple of months now. Summer vacation, became a father. I have less spare time now, but that doesn't mean this project is dead! I'm finishing up a development board I'll use for working with the smaller ATmega809/1609/3209, and expect to slowly catch up the next following weeks.

commit 0a860685 (2019-09-10): Change millis timer
Caused microcontroller to hang when TCB PWM outputs were modified. Closes #11
closed 2019-09-10 by MCUdude