MCUdude/MegaCoreX #13: Upload programmer does not seem to work for Curiosity Nano on Linux

egilkv, 2019-07-23:
Running the latest Arduino IDE 1.8.9 on a 64bit Ubuntu system.  Compiling to hex works fine, and I can program the Curiosity fine simply by copying the hex file to the mounted drive. Also the serial monitor works fine (using serial3). However, programming via the Arduino IDE (or running avrdude from the command line) I get this:
`         Using Port                    : usb
         Using Programmer              : curiosity_updi
avrdude: usbdev_open(): Found nEDBG CMSIS-DAP, serno: ATML3094051800000571
avrdude: usbdev_open(): WARNING: failed to set configuration 1: Device or resource busy
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
avrdude: usbdev_send(): wrote -5 out of 512 bytes, err = Input/output error
avrdude: jtag3_edbg_prepare(): failed to send command to serial port`

Avrdude command line is:
`/home/egil/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino14/bin/avrdude -C/home/egil/Arduino/hardware/MegaCoreX-master/megaavr/avrdude.conf -v -patmega4809 -ccuriosity_updi -Pusb -Uflash:w:/tmp/arduino_build_674501/blink_mega4809.ino.hex:i  `

I believe my system is fine, it works with everything from Attiny48 to ESP32, and I have made the udev rules for the USB port. Also, running avrdude sudo makes no difference. strace reveals the issue might be with some USBDEVFS_REAPURBNDELAY ioctls on the USB device. Would be very happy to contribute to fix the issue, but need some pointers on where to start,

MCUdude, 2019-09-15:
Sorry for the incredibly late reply. I don't have a Linux computer I can test with. What if you try to install the new 1.8.10 IDE together with the latest boards manager install for Uno Wifi Rev2/Nano Every? it uses a newer version of Avrdude.

egilkv, 2019-09-20:
Alas, latest IDE with avrdude Version 6.3-20190619 behaves the same for me. Also with sudo.

I will get a Curiosity Pro to see how that works in my environment.

MCUdude, 2019-09-22:
Sorry, but I'm no Avrdude guru at all. You should report the issue over at [Arduino's Avrdude fork](https://github.com/facchinm/avrdude), which seem like the most up-to-date version/fork of Avrdude.

Szum123321, 2021-05-13:
Same for me. Has anyone had any success with it?

MCUdude, 2022-01-12:
@dl8dtl is this an issue you can reproduce with the latest Avrdude build? I don't have a computer that runs Linux, but it works on Windows and Mac.

Related issue:
https://github.com/arduino/avrdude-build-script/issues/25

dl8dtl, 2022-01-12:
> @dl8dtl is this an issue you can reproduce with the latest Avrdude build? 

Seems to me the AVRDUDE binary has not been built with hidapi support. It falls back to using libusb then.

I guess it's time to replace the lengthy STK500 parameter printout on -v with a configuration summary instead, in order to diagnose such problems.

mcuee, 2022-10-16:
> I guess it's time to replace the lengthy STK500 parameter printout on -v with a configuration summary instead, in order to diagnose such problems.

Interesting idea. Indeed that may help.

MCUdude, 2023-01-10:
The latest MegaCoreX boards manager install have resolved this issue

closed 2023-01-10 by MCUdude