MCUdude/MegaCoreX #15: Most TCA0-based PWM outputs does not work

egilkv, 2019-07-26:
This is with the M4809 48 pin version. The two first potential outputs, 14 and 15, aka PC0 and PC1, are simply not enabled as potential outputs. I believe they should be, assuming Serial1 is not used. Output 16, PC2, seems to work just fine. Output 17, 18, 19, PC3 to PC5, are enabled as outputs, but the code is for 16 bit timer mode, not 8 bit split mode, so it simply does not work (writing to wrong registers, but seems benign).

I can contribute to fix this, but I need to know what is intended. To me it seems logical to use 8 bit mode, and allow PWM output on all six pins PC0 through PC5.

egilkv, 2019-07-29:
I have now implemented SPLIT mode for timer A, so all the promised PWM outputs now do actually work. Also, added support for PWM on the two pins assigned to Serial1, so you can either use these pins for Serial1 or for PWM output. The only thing missing is to update the pinout drawings to reflect PWM support also on pins PC0 and PC1. The change is here: https://github.com/MCUdude/MegaCoreX/pull/19

MCUdude, 2019-09-10:
Fixed in #19!

closed 2019-09-10 by MCUdude