MCUdude/MegaCoreX #24: attachInterrupt

Oaklands, 2019-07-29:
HI,
First of all a big thanks for providing this core support.

**Platform**: Curiosity Nano with ATmeg4809 @ 16Mhz

**Issue**: Cannot seem to get attachInterrupt running as expected on basic example (see below).

**Questions**: 
- Is there support for attachInterrupt on all pins or is it limited?
- Is attachInterrupt supported yet? Does it work the same way?
- What is the simplest way to service a GPIO external interrupt on this core

See **example test code** below (hopefully its not just a coding bug on my part)

Essentially I put an internal pullup on pin 36(PF2) and then short it to GND. However the interrupt does not seem to fire.

```
#define LED LED_BUILTIN
#define BUTTON 36
#define Serial Serial3 //Serial3 is the default on this core

volatile PinStatus state = LOW; //byte or uint8_t gives a compiler warning

void setup()
{
    Serial.begin(115200);

    pinMode(LED, OUTPUT);
    pinMode(BUTTON, INPUT_PULLUP);

    delay(10000);
    Serial.println("starting " __DATE__ " " __TIME__);

    attachInterrupt(BUTTON, blink, CHANGE);
    interrupts();
}

void loop()
{
    //Serial.println(digitalRead(BUTTON));
    digitalWrite(LED, state);
    //delay(1000);
}

void blink(void)
{
    state = digitalRead(BUTTON);
}
```

egilkv, 2019-07-30:
I have tested attachInterrupt(), and it worked for me. It should work on any pin, as far as I know. 

You do not mention which switch you are using. The built in switch is on PF6, Arduino pin 40. If you are connecting your own switch, you could verify that you have the correct pin by checking that the pullup is working. Also make sure you have selected the 48 pin variant.

Oaklands, 2019-07-30:
Thanks for the reply.

I have adjusted the code to use the button on the **Curiosity Nano ATMega4809**.
 
I also made the following changes:
- Changed the clock from 16Mhz to 4MHz (I didn't realize it was running at 3V3)
- Set Reset pin to GPIO using option under Tools menu (which works)
- IDE Compile settings: -fqbn=MegaCoreX:megaavr:4809:pinout=48pin_standard,resetpin=gpio,BOD=2v6,clock=4MHz_div -vid-pid=03EB_2175 -ide-version=10809 

I also tested using polling (instead of interrupts) and everything was OK.

I would appreciate if someone with the same board could test this sketch on their system to see if the LED goes ON/OFF as the button is pressed/released. (It doesn't work for me, nothing happens when I press/release the button.)

```
#define LED LED_BUILTIN
#define BUTTON 40  //button on Curiosity Nano (Reset Pin set to GPIO in IDE)
#define Serial Serial3

void setup()
{
    Serial.begin(115200);

    pinMode(LED, OUTPUT); //Led will default to ON
    pinMode(BUTTON, INPUT);

    delay(10000);
    Serial.println("starting " __DATE__ " " __TIME__);
    
    digitalWrite(LED,HIGH); //Turn LED OFF (after 10 sec delay above)

    attachInterrupt(BUTTON, blink, CHANGE);
    interrupts();
}

void loop()
{
    //do nothing here ...all done in ISR
}

void blink(void)
{ //when button is pressed LED is ON, OFF when button released
    if(digitalRead(BUTTON == HIGH))
        digitalWrite(LED, HIGH);
    else
        digitalWrite(LED, LOW);
}
```

MCUdude, 2019-09-15:
I can confirm that neither of the pins on PORTF work with external interrupts. I'll look into it. I'm sure there's a bug somewhere 😉

MCUdude, 2019-09-16:
Fixed in fe0da95!

closed 2019-09-16 by MCUdude

Oaklands, 2019-09-16:
Great....Many thanks for your efforts :)

I slightly modified the test sketch above and it works fine now - included below for reference.

**IDE settings:** Internal 4MHz, Reset as GPIO, Pinout 40/48 Standard, Programmer nEDBG

```
#define LED LED_BUILTIN
#define BUTTON 40  //button on Curiosity Nano (Reset Pin set to GPIO in IDE)
#define Serial Serial3

void setup()
{
  Serial.begin(115200);

  pinMode(LED, OUTPUT); //Led will default to ON
  pinMode(BUTTON, INPUT);

  delay(10000);
  Serial.println("starting " __DATE__ " " __TIME__);

  digitalWrite(LED, HIGH); //Turn LED OFF (after 10 sec delay above)

  attachInterrupt(BUTTON, blink, CHANGE);
  interrupts();
}

void loop()
{
  //do nothing here ...all done in ISR
}

void blink(void)
{ // Toggle the LED on every change in button status(pressed or not pressed)
  Serial.write('+');
  if (digitalRead(LED)) 
    digitalWrite(LED, LOW);
  else
      digitalWrite(LED, HIGH);
}
```