MCUdude/MegaCoreX #27: Lowest available baudrate for Serial is 2400

egilkv, 2019-08-02:
Anything less then 2400 is silently accepted, and results in some very strange actual speed. The Arduino documentation says anything down to 300 is OK, and with various radio technologies these baudrates are still relevant. This is a bug in the original megaavr code.

egilkv, 2019-08-02:
I have investigated, and is seems like this may be a limitation in the megaAVR baudrate generator, which is 16 bits with a fixed clock. Reducing the CPU frequency would extend the available range downwards.
This should at least be documented.

MCUdude, 2019-09-11:
Well, many projects out there relies on baud rates lower than 2400 baud, so we should try to fix this if possible. I still have to catch up on a lot of megaavr related code, but I'm happy to discuss with you and test on different hardware.

In the [datasheet](http://ww1.microchip.com/downloads/en/DeviceDoc/megaAVR-0-series-Family-Data-Sheet-40002015C.pdf) on page 79 I found this code, which apparently runs on 600 baud
```c
#include <assert.h>
 /* Baud rate compensated with factory stored frequency error */
 /* Asynchronous communication without Auto-baud (Sync Field) */
 /* 16MHz Clock, 3V and 600 BAUD */
 int8_t sigrow_val = SIGROW.OSC16ERR3V; // read signed error
 int32_t baud_reg_val = 600; // ideal BAUD register value

 assert (baud_reg_val >= 0x4A); // Verify legal min BAUD register value with max neg comp
 baud_reg_val *= (1024 + sigrow_val); // sum resolution + error
 baud_reg_val /= 1024; // divide by resolution
 USART0.BAUD = (int16_t) baud_reg_val; // set adjusted baud rate
```

MCUdude, 2019-09-15:
No, 600 is the preferred BAUD *register* value, not the actual baud rate. I don't think it's possible to achieve "super slow" baud rates with fast clocks.

commit 2241fb46 (2019-10-25): Add info about slow baud rates and fast clocks

Closes #27
closed 2019-10-25 by MCUdude