MCUdude/MegaCoreX #29: Support for Arduino Nano Every uploading

MCUdude, 2019-08-19:
This might be a tricky one. Since the new megaAVR-0 series relies on a separate programmer (mEDBG, nEDBG, JTAG2UPDI) I find it convenient that you can select your actual programming hardware under the Programmers menu. However, like Leonardo, the new Nano Every relies on `use_1200bps_touch=true` in order for the JTAG2UPDI interface to switch over to programming mode. I'm not sure how this can be achieved when the programming interface isn't defined in boards.txt, but rather in programmers.txt. @per1234 do you know if this is possible?

BTW I own an actual Nano Every, so I am able to do the testing.

MCUdude, 2019-08-21:
@facchinm do you know a way to trigger use_1200bps_touch from the programmers menu? It would be awesone if something like this would work.

``` 
jtag2updi.name=JTAG2UPDI
jtag2updi.communication=serial
jtag2updi.protocol=jtag2updi
jtag2updi.program.tool=avrdude
jtag2updi.program.extra_params=-P{serial.port}
jtag2updi.upload.use_1200bps_touch=true
```

facchinm, 2019-08-21:
Not right now but it could be a nice feature request for https://github.com/arduino/arduino-cli :wink:

MCUdude, 2019-08-21:
> Not right now but it could be a nice feature request for https://github.com/arduino/arduino-cli 😉

Sure, it belongs to arduino-cli too. I can create an issue if you want. So is this something that could be added to the IDE too? If so, it should be possible to use the SAMD11D14 to create a standalone programmer/serial adapter that can be selected in the programmers menu. Very similar to [microUPDI](https://github.com/MCUdude/microUPDI), but with open source software!

per1234, 2019-08-21:
The Arduino IDE will eventually use arduino-cli for everything except for its GUI elements. So when a feature is added to arduino-cli, it is also on track for being added to the Arduino IDE.

MCUdude, 2019-08-21:
> The Arduino IDE will eventually use arduino-cli for everything except for its GUI elements. So when a feature is added to arduino-cli, it is also on track for being added to the Arduino IDE.

Cool! Will it be compatible with existing boards.txt and programmers.txt files?

per1234, 2019-08-21:
Yes. arduino-cli is certainly being developed with the requirement of backwards compatibility with the existing hardware core system used in the Arduino IDE. They're essentially just moving all the existing non-GUI functionality of the Arduino IDE into a separate tool, and maybe adding in a few additional cool features along the way!

All IDEs can share the same tool instead of having to attempt to recreate all the functionality that was previously coded into the Arduino IDE every time. An improvement to arduino-cli will improve the Arduino IDE, Arduino Web Editor, and all the 3rd party IDEs using arduino-cli. And of course people who don't want an IDE can just use the much more lightweight arduino-cli alone.

MCUdude, 2019-10-26:
I got a clever idea last night on how this could be solved. The solution was pushed in 2a45bb5.
We now have Nano Every support!

closed 2019-10-26 by MCUdude

gordonthree, 2022-08-04:
sorry to reopen an old issue...

Any guidance on how I implement the *use_1200bps_touch* upload directive in platformio? I'm able to upload using MegaX in the Arduino IDE, but it PIO it fails to reset the com port from the default serial-passthrough mode. Somehow the native atmelmega core is doing it, but I haven't found where yet. This is for an official Arduino Nano Every.

```
avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/home/gordon/.platformio/packages/tool-avrdude-megaavr/avrdude.conf"
         User configuration file is "/home/gordon/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyACM0
         Using Programmer              : jtag2updi
         Overriding Baud Rate          : 115200
avrdude: jtagmkII_getsync(): sign-on command: status -1
```
that last line repeats forever until I ctrl-c

```
[env:nano_every]
platform = atmelmegaavr
framework = arduino
board = nano_every
board_build.f_cpu = 16000000L
board_hardware.oscillator = internal
build_unflags =
build_flags = -D MCU_NANOEVERY
              -D I2C_SLAVE_ADDR=0x39
			  -mrelax
			  -lprintf_flt
monitor_speed = 921600
monitor_port = /dev/ttyACM0
upload_protocol = jtag2updi
upload_port = /dev/ttyACM0
board_upload.speed = 115200
upload_flags = -v
```

MCUdude, 2022-08-04:
> sorry to reopen an old issue...
Any guidance on how I implement the use_1200bps_touch upload directive in platformio? I'm able to upload using MegaX in the Arduino IDE, but it PIO it fails to reset the com port from the default serial-passthrough mode. Somehow the native atmelmega core is doing it, but I haven't found where yet. This is for an official Arduino Nano Every.

@gordonthree you can add `board_upload.use_1200bps_touch = true`to your platformio.ini file