MCUdude/MegaCoreX #31: Can't compile on arm64 host

tgtakaoka, 2019-09-03:
On [Linux ARM 64 bits Arduino IDE 1.8.9](https://www.arduino.cc/download_handler.php?f=/arduino-1.8.9-linuxaarch64.tar.xz), compile sketch, for instance Examples/01. Basic/Blink, fails with the following message.

avr-g++: error: device-specs/specs-atmega4809: No such file or directory
exit status 1
Error compiling for board Atmega4809.

The same error happens for all target board of MegaCoreX.

Note that any targets of MCUdude/MegaCoreX works fine on [Linux 64 bits Arduino IDE 1.8.9](https://www.arduino.cc/download_handler.php?f=/arduino-1.8.9-linux64.tar.xz)host.
Both hosts are running Chrome OS + crouton + xiwi environment.

Also note that any targets of [SpenceKonde/megaTinyCore](https://github.com/SpenceKonde/megaTinyCore) works fine on Linux ARM 64 bits host.

MCUdude, 2019-09-03:
@per1234 is this a known problem?

tgtakaoka, 2019-09-13:
FYI: Here is detailed log message while compiling.

```
Arduino: 1.8.9 (Linux), Board: "Atmega4809, Uno WiFi, Reset, BOD 2.6V, 16 MHz"

/home/t2/.local/arduino-1.8.9/arduino-builder -dump-prefs -logger=machine -hardware /home/t2/.local/arduino-1.8.9/hardware -hardware /home/t2/.arduino15/packages -hardware /home/t2/Arduino/hardware -tools /home/t2/.local/arduino-1.8.9/tools-builder -tools /home/t2/.local/arduino-1.8.9/hardware/tools/avr -tools /home/t2/.arduino15/packages -built-in-libraries /home/t2/.local/arduino-1.8.9/libraries -libraries /home/t2/Arduino/libraries -fqbn=MegaCoreX:megaavr:4809:pinout=uno_wifi,resetpin=reset,BOD=2v6,clock=16MHz -ide-version=10809 -build-path /tmp/arduino_build_440953 -warnings=default -build-cache /tmp/arduino_cache_741524 -prefs=build.warn_data_percentage=75 -verbose /home/t2/.local/arduino-1.8.9/examples/01.Basics/Blink/Blink.ino
/home/t2/.local/arduino-1.8.9/arduino-builder -compile -logger=machine -hardware /home/t2/.local/arduino-1.8.9/hardware -hardware /home/t2/.arduino15/packages -hardware /home/t2/Arduino/hardware -tools /home/t2/.local/arduino-1.8.9/tools-builder -tools /home/t2/.local/arduino-1.8.9/hardware/tools/avr -tools /home/t2/.arduino15/packages -built-in-libraries /home/t2/.local/arduino-1.8.9/libraries -libraries /home/t2/Arduino/libraries -fqbn=MegaCoreX:megaavr:4809:pinout=uno_wifi,resetpin=reset,BOD=2v6,clock=16MHz -ide-version=10809 -build-path /tmp/arduino_build_440953 -warnings=default -build-cache /tmp/arduino_cache_741524 -prefs=build.warn_data_percentage=75 -verbose /home/t2/.local/arduino-1.8.9/examples/01.Basics/Blink/Blink.ino
Using board '4809' from platform in folder: /home/t2/Arduino/hardware/MegaCoreX/megaavr
Using core 'coreX-corefiles' from platform in folder: /home/t2/Arduino/hardware/MegaCoreX/megaavr
Detecting libraries used...
/home/t2/.local/arduino-1.8.9/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_ATmega4809 -DARDUINO_ARCH_MEGAAVR -DUNO_WIFI_REV2_328MODE -I/home/t2/Arduino/hardware/MegaCoreX/megaavr/cores/coreX-corefiles/api/deprecated -I/home/t2/Arduino/hardware/MegaCoreX/megaavr/cores/coreX-corefiles -I/home/t2/Arduino/hardware/MegaCoreX/megaavr/variants/uno-wifi /tmp/arduino_build_440953/sketch/Blink.ino.cpp -o /dev/null
Error while detecting libraries included by /tmp/arduino_build_440953/sketch/Blink.ino.cpp
Generating function prototypes...
/home/t2/.local/arduino-1.8.9/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_ATmega4809 -DARDUINO_ARCH_MEGAAVR -DUNO_WIFI_REV2_328MODE -I/home/t2/Arduino/hardware/MegaCoreX/megaavr/cores/coreX-corefiles/api/deprecated -I/home/t2/Arduino/hardware/MegaCoreX/megaavr/cores/coreX-corefiles -I/home/t2/Arduino/hardware/MegaCoreX/megaavr/variants/uno-wifi /tmp/arduino_build_440953/sketch/Blink.ino.cpp -o /tmp/arduino_build_440953/preproc/ctags_target_for_gcc_minus_e.cpp
avr-g++: error: device-specs/specs-atmega4809: No such file or directory
exit status 1
Error compiling for board Atmega4809.
```

MCUdude, 2019-09-13:
What happens if you install the Uno Wifi/Nano every boards manager option?

MCUdude, 2019-09-13:
Never mind. I downloaded IDE 1.8.9 ARM64 just to look at its contents. It's true that `device-specs/specs-atmega4809` is missing. However, that's not MegaCoreX's fault but rather the IDE. The file has recently been added, so download and install the latest IDE hourly release instead!

tgtakaoka, 2019-09-13:
Hi, thank you for the info.
I downloaded the 1.8.9 ARM64 IDE just now and extracted the content, bit can't find "specs-atmega4809" in it (the content are sightly different that I had installed previously).

$ md5sum ~/Downloads/*.tar.xz
03b0ca699cad1307f5b1c1d1a6deb614  /home/t2/Downloads/arduino-1.8.9-linuxaarch64.tar.xz

And I also confirmed that the same error message was reported while compiling Blink targeting to ATmega4809.

I understand the root cause is missing files in Arduino IDE. I'll keep watching it.

Do you mind keeping this issue open until problem will be solved, just for the record?

MCUdude, 2019-09-13:
> Do you mind keeping this issue open until the problem will be solved, just for the record?

Sure, I'll close it when IDE 1.8.10 is released!

tgtakaoka, 2019-09-13:
Wow, Arduino IDE 1.8.10 is just released.
And with master tip-of-toe MCUdude/MegaCoreX and 1.8.10, compiling sketch targeting to ATmega4809 success with no error!

/home/t2/.local/arduino-1.8.10/arduino-builder -dump-prefs -logger=machine -hardware /home/t2/.local/arduino-1.8.10/hardware -hardware /home/t2/.arduino15/packages -hardware /home/t2/Arduino/hardware -tools /home/t2/.local/arduino-1.8.10/tools-builder -tools /home/t2/.local/arduino-1.8.10/hardware/tools/avr -tools /home/t2/.arduino15/packages -built-in-libraries /home/t2/.local/arduino-1.8.10/libraries -libraries /home/t2/Arduino/libraries -fqbn=MegaCoreX:megaavr:4809:pinout=48pin_standard,resetpin=reset,BOD=2v6,clock=internal_16MHz -ide-version=10810 -build-path /tmp/arduino_build_165025 -warnings=default -build-cache /tmp/arduino_cache_880990 -prefs=build.warn_data_percentage=75 -verbose /home/t2/.local/arduino-1.8.10/examples/01.Basics/Blink/Blink.ino
/home/t2/.local/arduino-1.8.10/arduino-builder -compile -logger=machine -hardware /home/t2/.local/arduino-1.8.10/hardware -hardware /home/t2/.arduino15/packages -hardware /home/t2/Arduino/hardware -tools /home/t2/.local/arduino-1.8.10/tools-builder -tools /home/t2/.local/arduino-1.8.10/hardware/tools/avr -tools /home/t2/.arduino15/packages -built-in-libraries /home/t2/.local/arduino-1.8.10/libraries -libraries /home/t2/Arduino/libraries -fqbn=MegaCoreX:megaavr:4809:pinout=48pin_standard,resetpin=reset,BOD=2v6,clock=internal_16MHz -ide-version=10810 -build-path /tmp/arduino_build_165025 -warnings=default -build-cache /tmp/arduino_cache_880990 -prefs=build.warn_data_percentage=75 -verbose /home/t2/.local/arduino-1.8.10/examples/01.Basics/Blink/Blink.ino
Using board '4809' from platform in folder: /home/t2/Arduino/hardware/MegaCoreX/megaavr
Using core 'coreX-corefiles' from platform in folder: /home/t2/Arduino/hardware/MegaCoreX/megaavr
Detecting libraries used...
/home/t2/.local/arduino-1.8.10/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_ATmega4809 -DARDUINO_ARCH_MEGAAVR -I/home/t2/Arduino/hardware/MegaCoreX/megaavr/cores/coreX-corefiles/api/deprecated -I/home/t2/Arduino/hardware/MegaCoreX/megaavr/cores/coreX-corefiles -I/home/t2/Arduino/hardware/MegaCoreX/megaavr/variants/48pin-standard /tmp/arduino_build_165025/sketch/Blink.ino.cpp -o /dev/null
Generating function prototypes...
/home/t2/.local/arduino-1.8.10/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_ATmega4809 -DARDUINO_ARCH_MEGAAVR -I/home/t2/Arduino/hardware/MegaCoreX/megaavr/cores/coreX-corefiles/api/deprecated -I/home/t2/Arduino/hardware/MegaCoreX/megaavr/cores/coreX-corefiles -I/home/t2/Arduino/hardware/MegaCoreX/megaavr/variants/48pin-standard /tmp/arduino_build_165025/sketch/Blink.ino.cpp -o /tmp/arduino_build_165025/preproc/ctags_target_for_gcc_minus_e.cpp
/home/t2/.local/arduino-1.8.10/tools-builder/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /tmp/arduino_build_165025/preproc/ctags_target_for_gcc_minus_e.cpp
Compiling sketch...
/home/t2/.local/arduino-1.8.10/hardware/tools/avr/bin/avr-g++ -c -g -Os -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_ATmega4809 -DARDUINO_ARCH_MEGAAVR -I/home/t2/Arduino/hardware/MegaCoreX/megaavr/cores/coreX-corefiles/api/deprecated -I/home/t2/Arduino/hardware/MegaCoreX/megaavr/cores/coreX-corefiles -I/home/t2/Arduino/hardware/MegaCoreX/megaavr/variants/48pin-standard /tmp/arduino_build_165025/sketch/Blink.ino.cpp -o /tmp/arduino_build_165025/sketch/Blink.ino.cpp.o
Compiling libraries...
Compiling core...
/home/t2/.local/arduino-1.8.10/hardware/tools/avr/bin/avr-gcc -c -g -Os -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_ATmega4809 -DARDUINO_ARCH_MEGAAVR -I/home/t2/Arduino/hardware/MegaCoreX/megaavr/cores/coreX-corefiles/api/deprecated -I/home/t2/Arduino/hardware/MegaCoreX/megaavr/cores/coreX-corefiles -I/home/t2/Arduino/hardware/MegaCoreX/megaavr/variants/48pin-standard /home/t2/Arduino/hardware/MegaCoreX/megaavr/variants/48pin-standard/variant.c -o /tmp/arduino_build_165025/core/variant.c.o
/home/t2/.local/arduino-1.8.10/hardware/tools/avr/bin/avr-gcc -c -g -x assembler-with-cpp -flto -MMD -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_ATmega4809 -DARDUINO_ARCH_MEGAAVR -I/home/t2/Arduino/hardware/MegaCoreX/megaavr/cores/coreX-corefiles/api/deprecated -I/home/t2/Arduino/hardware/MegaCoreX/megaavr/cores/coreX-corefiles -I/home/t2/Arduino/hardware/MegaCoreX/megaavr/variants/48pin-standard /home/t2/Arduino/hardware/MegaCoreX/megaavr/cores/coreX-corefiles/wiring_pulse.S -o /tmp/arduino_build_165025/core/wiring_pulse.S.o
/home/t2/.local/arduino-1.8.10/hardware/tools/avr/bin/avr-gcc -c -g -Os -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_ATmega4809 -DARDUINO_ARCH_MEGAAVR [... cut ...] arduino_build_165025/Blink.ino.eep
/home/t2/.local/arduino-1.8.10/hardware/tools/avr/bin/avr-objcopy -O ihex -R .eeprom /tmp/arduino_build_165025/Blink.ino.elf /tmp/arduino_build_165025/Blink.ino.hex
/home/t2/.local/arduino-1.8.10/hardware/tools/avr/bin/avr-size -A /tmp/arduino_build_165025/Blink.ino.elf
Sketch uses 1380 bytes (2%) of program storage space. Maximum is 49152 bytes.
Global variables use 18 bytes (0%) of dynamic memory, leaving 6126 bytes for local variables. Maximum is 6144 bytes.

MCUdude, 2019-09-13:
What a coincidence that IDE 1.8.10 was released today! Great to hear it's working fine.

closed 2019-09-13 by MCUdude