MCUdude/MegaCoreX #32: ATmega4809-P DIP-40 target reports fuse5 warning when uploading sketch

tgtakaoka, 2019-09-03:
Compiling Examples>01.Basic>Blink as targeting MegaCoreX/ATmega4809/48 pin standard works fine, but reports a warning related to unused bit in fuse5.
Programmer option is Atmel mEDBG (ATmega32u4), and the hardware is an awesome microUPDI of course :).

I'm not sure this is DIP-40 related or not, since I have no 48 pins 4809 in hand.

Global variables use 22 bytes (0%) of dynamic memory, leaving 6122 bytes for local variables. Maximum is 6144 bytes.
avrdude: usbdev_open(): WARNING: failed to set configuration 1: Device or resource busy
avrdude: WARNING: invalid value for unused bits in fuse "fuse5", should be set to 1 according to datasheet
This behaviour is deprecated and will result in an error in future version
You probably want to use 0xcd instead of 0xc9 (double check with your datasheet first).

MCUdude, 2019-09-03:
Thanks for reporting. It's not related to the DIP-40 package, it's caused by the fuse bits settings.
The error message is right; the SYSCFG0 byte should be 0xCD instead of 0xC9. It's not critical at all, but I will fix it to prevent this warning from appearing.

Out of curiosity, did you order yourself a microUPDI board from OSHpark or another PCB manufacturer? And what does your DIP-40 board look like? 🙂

tgtakaoka, 2019-09-03:
I created my own using a Pro Micro following your instruction at microUPDI. Thanks a lot!
My ATmega4809-P board is just a chip and bread board.

[image]

tgtakaoka, 2019-09-03:
And it will be great if we can have 40 pin standard pinout option.

MCUdude, 2019-09-03:
I do have a pinout diagram for the 4809 DIP40 I haven't uploaded yet. However, I don't think I'm going to create a dedicated pinout for the DIP-40 variant because inconvenient to have too many pinouts for the same chip. By using the **Standard pinout** with the DIP40, you'll lose digital pin 8 to 13 + 20 and 21. Apart from this everything is like before.

commit 4a809249 (2019-09-10): Fix unset fuse warning
This warning appears only on the microUPDI/mEDBG. No idea why. Closes #32
closed 2019-09-10 by MCUdude