MCUdude/MegaCoreX #33: Millis/micros not accurate for all clock frequencies

MCUdude, 2019-09-14:
It turns out that millis() and micros aren't accurate at all if we're *not* running at 8 MHz or 16 MHz.

Something has to be done because millis/micros are pretty much useless at the moment.
If someone wants to participate, here's the test sketch I'm using. An oscilloscope is used to read the output of PB2.

```c++
uint32_t previousMillis = 0;
uint32_t currentMillis = 0;

const uint16_t interval = 50;

void setup() 
{
  PORTB.DIR |= PIN2_bm; // PB2 as output
}

void loop() 
{
  while(1)
  {
    uint32_t currentMillis = micros();

    if (currentMillis - previousMillis >= interval) 
    {
      previousMillis = currentMillis;
      PORTB.OUTTGL = PIN2_bm; // Toggle PB2
    }
  }
}
```

MCUdude, 2019-09-15:
Commit e4438ca greatly improves accuracy for all relevant clocks. I'm still experiencing a lot of jitter when working with 4 MHz or lower clocks, especially when I try to acheive pulses shorter than 100us with the code above. Most pulses are spot on, some are too slow, but supprisingly, some are too fast. I don't understand what's causing this really. BTW you need a digital oscilloscope to capture this. @SpenceKonde I'm pretty sure this applies for **megaTinyCore** too.

MCUdude, 2019-09-15:
Here's micros function, slightly modified compared the upstream one. I've added a section for a 4 MHz clock because that's what I'm working on at the moment.

``` c
unsigned long  micros() {
	uint32_t m;
	uint8_t t;

	/* Save current state and disable interrupts */
	uint8_t status = SREG;
	cli();

	/* Get current number of overflows and timer count */
	m = timer_overflow_count;
	t = _timer->CNTL;

	/* If the timer overflow flag is raised, we just missed it,
	increment to account for it, & read new ticks */
	if(_timer->INTFLAGS & TCB_CAPT_bm){
		m++;
		t = _timer->CNTL;
	}

	// Restore SREG
	SREG = status;

#if F_CPU >= 24000000L && F_CPU < 32000000L
	// m needs to be multiplied by 682.67
	// and t by 2.67
	m = (m << 8) + t;
	return (m << 1) + (m >> 1) + (m >> 3) + (m >> 4); // Multiply by 2.6875
#elif F_CPU == 20000000L
	// m needs to be multiplied by 819.2
	// t needs to be multiplied by 3.2
	m = (m << 8) + t;
	return m + (m << 1) + (m >> 2) - (m >> 4); // Multiply by 3.1875
#elif F_CPU == 12000000L
	// m needs to be multiplied by 1365.33
	// and t by 5.33
	m = (m << 8) + t;
	return m + (m << 2) + (m >> 2) + (m >> 3) - (m >> 4) + (m >> 5); // Multiply by 5.3437
#elif F_CPU == 4000000L
  m = (m << 8) + t;
  return (m << 4);
#else
	return ((m << 8)) * (64 / clockCyclesPerMicrosecond());
#endif
```

Apparently, it the addition of `t` that screws up everything. I'm not sure what's going on, but on my scope screen it seems like it varies how many clock cycles it takes to add `t` to `m`.

By removing `+ t` the pulses get much longer, but still, have a lot less jitter, and starts to behave like the old AVRs did. The pulse may be longer due to interrupts, but never shorter.

@nerdralph I know you know your assembly magic. Is there a way to convert: 
``` c
#elif F_CPU == 4000000L
  m = (m << 8);
  return (m << 4);
```

or even better (this is more accurate):
``` c
#elif F_CPU == 4000000L
  m = (m << 8);
  return (m << 1) + (m << 4);
```

Into some inline assembly that we know takes a given amount of clock cycles to execute?

Finally, Here are some screenshots from my scope. I'm trying to acheive a pulsewidth of 50us.
[image]
[image]

nerdralph, 2019-09-15:
Is the init() code in wiring.c still the same?  If so, then the prescaler
is 8, so at 4Mhz timer/counter0 will increment every 2us.  Overflows will
occur every 512us, so I'd expect the micros() code to be:
return (m << 9) + (t<<1);
The code should be deterministic.  I'll do some tests later to see how well
the compiler optimizes it.
I'd like to see your code that is looping to check micros().
You should be able to improve the speed of micros() and the isr by using
__uint24 instead of uint32.  Since the micros() function always multiplies
by more than 256, the upper 8 bits of timer0_overflow are never used.
The non-deterministic part of micros() that will cause some jitter is near
the start of the code.

if(_timer->INTFLAGS & TCB_CAPT_bm)

nerdralph, 2019-09-16:
I finished doing some tests with avr-gcc 4.9.2, and it does a poor job of
optimizing the math.
return ((m<<8) + t)<<1;
generates smaller code.  It's still not the smallest possible code, but
it's not bad, and as expected it is deterministic (always the same number
of cycles to execute).

MCUdude, 2019-09-17:
Thank you @nerdralph! I'll do some more testing the following days and report back.

egilkv, 2019-09-24:
Hopefully micros/millis are much better now, with the new timer implementation. Should also work on all clocks. 

Do note that with interrupts enabled, you will experience some jitter due to interrupt routines, if nothing else. If you want something totally jitter free, you have to make a timer control an output directly, which you really have to do yourself.

MCUdude, 2019-09-24:
> Do note that with interrupts enabled, you will experience some jitter due to interrupt routines, if nothing else. If you want something totally jitter free, you have to make a timer control an output directly, which you really have to do yourself.

We also have lots of jitter on the older AVRs too. At least now the reported micros duration is never shorter than the actual duration.