MCUdude/MegaCoreX #34: Avrdude does not work with Arduino 1.8.10

egilkv, 2019-09-20:
Trivial fix in megaavr/platform.txt:

tools.avrdude.path={runtime.tools.avrdude-6.3.0-arduino17.path}

MCUdude, 2019-09-20:
What error do you get? I prefer not to hard-code the Avrdude version every time there's a new release either.

egilkv, 2019-09-20:
Did not note down exactly, but could definitely not find avrdude, 
probably because of undefined macro. No idea if there is a suitable 
version independent variant.

MCUdude, 2019-09-22:
No issues here.
This is the output (on Mac) when building the blink sketch on a 100% clean Arduino 1.8.10 install without any boards manager packages installed:

```
/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/hans/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/hans/Documents/Arduino/libraries -fqbn=MegaCoreX:megaavr:4809:pinout=48pin_standard,resetpin=reset,BOD=2v6,clock=internal_16MHz -ide-version=10810 -build-path /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_849639 -warnings=all -build-cache /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_cache_763525 -prefs=build.warn_data_percentage=75 -verbose /Applications/Arduino.app/Contents/Java/examples/01.Basics/Blink/Blink.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/hans/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/hans/Documents/Arduino/libraries -fqbn=MegaCoreX:megaavr:4809:pinout=48pin_standard,resetpin=reset,BOD=2v6,clock=internal_16MHz -ide-version=10810 -build-path /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_849639 -warnings=all -build-cache /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_cache_763525 -prefs=build.warn_data_percentage=75 -verbose /Applications/Arduino.app/Contents/Java/examples/01.Basics/Blink/Blink.ino
Using board '4809' from platform in folder: /Users/hans/Documents/Arduino/hardware/MegaCoreX/megaavr
Using core 'coreX-corefiles' from platform in folder: /Users/hans/Documents/Arduino/hardware/MegaCoreX/megaavr
Detecting libraries used...
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_ATmega4809 -DARDUINO_ARCH_MEGAAVR -I/Users/hans/Documents/Arduino/hardware/MegaCoreX/megaavr/cores/coreX-corefiles/api/deprecated -I/Users/hans/Documents/Arduino/hardware/MegaCoreX/megaavr/cores/coreX-corefiles -I/Users/hans/Documents/Arduino/hardware/MegaCoreX/megaavr/variants/48pin-standard /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_849639/sketch/Blink.ino.cpp -o /dev/null
Generating function prototypes...
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_ATmega4809 -DARDUINO_ARCH_MEGAAVR -I/Users/hans/Documents/Arduino/hardware/MegaCoreX/megaavr/cores/coreX-corefiles/api/deprecated -I/Users/hans/Documents/Arduino/hardware/MegaCoreX/megaavr/cores/coreX-corefiles -I/Users/hans/Documents/Arduino/hardware/MegaCoreX/megaavr/variants/48pin-standard /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_849639/sketch/Blink.ino.cpp -o /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_849639/preproc/ctags_target_for_gcc_minus_e.cpp
/Applications/Arduino.app/Contents/Java/tools-builder/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_849639/preproc/ctags_target_for_gcc_minus_e.cpp
Compiling sketch...
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_ATmega4809 -DARDUINO_ARCH_MEGAAVR -I/Users/hans/Documents/Arduino/hardware/MegaCoreX/megaavr/cores/coreX-corefiles/api/deprecated -I/Users/hans/Documents/Arduino/hardware/MegaCoreX/megaavr/cores/coreX-corefiles -I/Users/hans/Documents/Arduino/hardware/MegaCoreX/megaavr/variants/48pin-standard /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_849639/sketch/Blink.ino.cpp -o /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_849639/sketch/Blink.ino.cpp.o
Compiling libraries...
Compiling core...
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc -c -g -Os -Wall -Wextra -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_ATmega4809 -DARDUINO_ARCH_MEGAAVR -I/Users/hans/Documents/Arduino/hardware/MegaCoreX/megaavr/cores/coreX-corefiles/api/deprecated -I/Users/hans/Documents/Arduino/hardware/MegaCoreX/megaavr/cores/coreX-corefiles -I/Users/hans/Documents/Arduino/hardware/MegaCoreX/megaavr/variants/48pin-standard /Users/hans/Documents/Arduino/hardware/MegaCoreX/megaavr/variants/48pin-standard/variant.c -o /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_849639/core/variant.c.o
[... cut ...]
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-objcopy -O ihex -R .eeprom /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_849639/Blink.ino.elf /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_849639/Blink.ino.hex
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-size -A /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_849639/Blink.ino.elf
Sketch uses 1006 bytes (2%) of program storage space. Maximum is 49152 bytes.
Global variables use 18 bytes (0%) of dynamic memory, leaving 6126 bytes for local variables. Maximum is 6144 bytes.
```

egilkv, 2019-09-22:
I see no avrdude in your log, I see only the compiler and linker etc. The problem for me was the upload step, i.e. using avrdude. I did install the new Arduino using install.sh, which I assume removes the previous avrdude version.

MCUdude, 2019-09-22:
You're right. The Avrdude path is fixed in db73134

egilkv, 2019-09-22:
Excellent!

closed 2019-09-22 by MCUdude