MCUdude/MegaCoreX #35: NEW PINMAP

cheny3922, 2019-09-21:
I'd like to redesign pinmap，As compatible as possible UNO and MEGA2560,  D0-D1:SERIAL ;  D2-D11 HAVE PWM,  D14-D19:SERIAL;  D10-D14:SPI; 
See the table below for details：
NUM | PIN | PWM | UART | IIC | SPI | ADC |  
-- | -- | -- | -- | -- | -- | -- | --
0 | PA0 |   | T0 |   |   |   |  
1 | PA1 |   | R0 |   |   |   |  
2 | PA2 | TCB0 |   | SDA |   |   |  
3 | PA3 | TCB1 |   | SCL |   |   |  
4 | PB0 | TCA | T3 |   |   |   |  
5 | PB1 | TCA | R3 |   |   |   |  
6 | PB2 | TCA |   |   |   |   |  
7 | PB3 | TCA |   |   |   |   |  
8 | PB4 | TCA | T3 |   |   |   |  
9 | PB5 | TCA | R3 |   |   |   |  
10 | PC0 | TCB2 | T1 |   | MO |   |  
11 | PC1 | TCB3 | R1 |   | MI |   |  
12 | PC2 |   |   | SDA | SCK |   |  
13 | PC3 |   |   | SCL | SS |   |  
14 | PF0 |   | T2 |   |   |   |  
15 | PF1 |   | R2 |   |   |   |  
16 | PC4 |   | T1 |   |   |   |  
17 | PC5 |   | R1 |   |   |   |  
18 | PA4 |   | T0 |   | MO |   |  
19 | PA5 |   | R0 |   | MI |   |  
20 | PA6 |   |   |   | SCK |   |  
21 | PA7 |   |   |   | SS |   |  
22 | PC6 |   |   |   |   |   |  
23 | PC7 |   |   |   |   |   |  
24 | PD0 |   |   |   |   | A0 |  
25 | PD1 |   |   |   |   | A1 |  
26 | PD2 |   |   |   |   | A2 |  
27 | PD3 |   |   |   |   | A3 |  
28 | PD4 |   |   |   |   | A4 |  
29 | PD5 |   |   |   |   | A5 |  
30 | PD6 |   |   |   |   | A6 |  
31 | PD7 |   |   |   |   | A7 |  
32 | PE0 |   |   |   | MO | A8 |  
33 | PE1 |   |   |   | MI | A9 |  
34 | PE2 |   |   |   | SCK | A10 |  
35 | PE3 |   |   |   | SS | A11 |  
36 | PF2 |   |   | SDA |   | A12 |  
37 | PF3 |   |   | SCL |   | A13 |  
38 | PF4 | TCB1 | T2 |   |   | A14 |  
39 | PF5 | TCB1 | R2 |   |   | A15 |  
40 | PF6 |   |   |   |   |   | RESET

cheny3922, 2019-09-21:
exchange  PA0-D1-TXD0,PA1-D0-RXD0,PB5-D8-RXD3,PB4-D9-TXD3

MCUdude, 2019-10-17:
In my opinion, we don't need more pin mappings than we already have. Feel free to fork this repo and implement it yourself if this is important for something you're working on.

closed 2019-10-17 by MCUdude