MCUdude/MegaCoreX #39: "upload" pattern problems.

WestfW, 2019-10-04:
The "upload" pattern in platform.txt contains both "-e" (erase chip) and "-D" (disable flash auto-erase.)  This is probably wrong.

I would rather than one of these be passed upward in "upload.extra_params" so that the boards.txt can select between bootloader (needs -D) and programmer (probably needs -e)

Also, the pattern includes the -b{upload.speed} options, but not the -P{serial.port}

MCUdude, 2019-10-04:
To be honest, most of the contents in platform.txt is just an old copy of the one from _ArduinoCore-megaavr_. And since none of the targets uses a bootloader, I've made it so that when you click _Upload_ it triggers the `program` part of platform.txt.

Later, when Optiboot is implemented it will be a dedicated menu option that will tell the IDE to use `upload` instead of `program`. Similar to what I've done with my other Arduino cores, where using a bootloader is optional.

I suggest changing [line 105](https://github.com/MCUdude/MegaCoreX/blob/a7bdb9b75a3e46fd121320a8cf89a54823f15db5/megaavr/platform.txt#L105) to
```
tools.avrdude.upload.pattern="{cmd.path}" "-C{config.path}" {upload.verbose} {upload.verify} -p{build.mcu} -c{upload.protocol} {upload.extra_params} -P{serial.port} -b{upload.speed} -D "-Uflash:w:{build.path}/{build.project_name}.hex:i"
```

MCUdude, 2019-10-06:
Nothing? No comment?

WestfW, 2019-10-07:
I'm still thinking.  The last couple of days have been spent chasing (apparent) Atmel Studio and/or Xplained dev board problems :-(
I wonder if -D is supposed to enable per-page erase before program (on xMega), or just the automatic chip erase?  Optiboot will always erase the page before programming, but at least theoretically, you could be flipping bits from 1's to 0's only if you write a page without erasing it first.

MCUdude, 2019-10-25:
I'm closing this issue because the current upload pattern seems to work fine. Please reopen if you think otherwise.

closed 2019-10-25 by MCUdude