MCUdude/MegaCoreX #43: delay/millis/micros don't work on ATmega808/809

MCUdude, 2019-10-08:
For some mysterious reason delay/millis/micros don't work on ATmega808/809. Calling in of these functions causes the microcontroller to hang. It does not happen on ATmega1609. I've tried to change the millis generation timer without any improvements. Any idea what this might be @egilkv ?

MCUdude, 2019-10-08:
I tried to load a file compiled for the ATmega1609 into an ATmega809, and it works now. Maybe there's a bug in iom808.h/iom809.h?

MCUdude, 2019-10-08:
None of the PWM output pins work either.

egilkv, 2019-10-09:
No experience with mega809. But if another variant works, then I would definitely look at the header definitions. 

For the original implementation, all PWMs as well as the millis would run from the same TCA0 divided down clock.

In my new implementation all PWMs still uses the same divided-down TCA0 base, while the TCBx used for millis() is independent.

MCUdude, 2019-10-09:
It looks like some of the device spec files are missing (or placed in the wrong folder). There are no significant differences between iom809.h and iom1609.h apart for some macros that define flash and ram size. I'm not sure who I should contact in order to figure this out...

WestfW, 2019-12-10:
There is some "discussion" on arvfreaks recently about whether the mega809 has 16bit or 32bit interrupt vectors...

https://www.avrfreaks.net/forum/atmega809-interrupt-vectors-datasheet-error

(if in fact the pack and the datasheet are wrong, then the timer interrupt will be going to the wrong place, and millis will never increment!)

MCUdude, 2019-12-10:
Interesting! Clearly these two issues are related.

SpenceKonde, 2020-05-19:
Yup - just confirmed with my modified compiler toolchain package (based on the new atpacks) that it does generate the 4-byte vector table (pathetic that they let something like this slip through - the 808 really *ought* to be using 2-byte vectors, so-as not to waste the very limited flash on that part)

MCUdude, 2020-06-30:
Fixed in the latest Arduino IDE 1.8.13 toolchain and/or the latest megaavr package

closed 2020-06-30 by MCUdude