MCUdude/MegaCoreX #47: Logic library support for tinyAVR-0,1 series

tgtakaoka, 2019-10-10:
Current Logic library supports only megaAVR-0 series. It would be great if Logic library can support tinyAVR-0 and tinyAVR-1 series as well.

MCUdude, 2019-10-10:
I agree that it's better to discuss Logic library support in a separate issue.

> I checked datasheets of ATtiny1607 and ATtiny3217 and realized that those have only 2 CCL blocks. Also those has different input/output mapping, even different port for alternate output :(.

We would have to take every "family" into account. That means that these four families will have to be handled properly.
"2 series" ATtiny202/212/204/404
"4 series" ATtiny204/214/404/414/804/814/1604/1614/3204/3214
"6 series" ATtiny406/416/806/816/1606/1616/3206/3216
"7 series" ATtiny407/417/807/817/1607/1617/3207/3217

Make sure you don't just Google-search to find the appropriate datasheet. Instead, fing the datasheet on Microchip's website. There's a lot of older datasheets floating around (copyrighted 2017) that have tons of errors. The new ones are much more consistent.

tgtakaoka, 2019-10-10:
This is [prototype](https://github.com/tgtakaoka/MegaCoreX/tree/support-tinyavr) that supports all tinyAVR-0,1 series. All examples except Interrupt can be successfully compiled using [megaTinyCore](https://github.com/SpenceKonde/megaTinyCore).

WDYT?

MCUdude, 2019-10-10:
Looks great! I just left for vacation, so I'm not near a computer before evening tomorrow, Norwegian time.

Hos is the memory usage compared to the non-attiny one?

MCUdude, 2019-10-10:
OK, I couldn't wait. I compiled the library, and as expected it uses a bit more flash, but that's OK. 

I get this compiler warning though. Why redefine CCL_OUTEN_bm? If we have to for some reason, maybe we should call it something else to prevent a compiler warning?
```
/Users/hans/Documents/Arduino/hardware/MegaCoreX/megaavr/libraries/Logic/src/Logic.cpp:102:0: warning: "CCL_OUTEN_bm" redefined
 #define CCL_OUTEN_bm _BV(6)
 
In file included from /Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include/avr/io.h:677:0,
                 from /Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include/avr/pgmspace.h:90,
                 from /Users/hans/Documents/Arduino/hardware/MegaCoreX/megaavr/cores/coreX-corefiles/api/String.h:30,
                 from /Users/hans/Documents/Arduino/hardware/MegaCoreX/megaavr/cores/coreX-corefiles/api/Print.h:24,
                 from /Users/hans/Documents/Arduino/hardware/MegaCoreX/megaavr/cores/coreX-corefiles/api/Stream.h:25,
                 from /Users/hans/Documents/Arduino/hardware/MegaCoreX/megaavr/cores/coreX-corefiles/api/Client.h:22,
                 from /Users/hans/Documents/Arduino/hardware/MegaCoreX/megaavr/cores/coreX-corefiles/api/ArduinoAPI.h:29,
                 from /Users/hans/Documents/Arduino/hardware/MegaCoreX/megaavr/cores/coreX-corefiles/Arduino.h:23,
                 from /Users/hans/Documents/Arduino/hardware/MegaCoreX/megaavr/libraries/Logic/src/Logic.h:4,
                 from /Users/hans/Documents/Arduino/hardware/MegaCoreX/megaavr/libraries/Logic/src/Logic.cpp:1:
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include/avr/iom4809.h:3226:0: note: this is the location of the previous definition
 #define CCL_OUTEN_bm  0x40  /* Output Enable bit mask. */
```

tgtakaoka, 2019-10-11:
Thank you for pointing out. I didn't realized that CCL_OUTEN_bm is defined in io header.

I request https://github.com/MCUdude/MegaCoreX/pull/50 to implement this feature request.

- declares struct Logic::CCLBlock as forward declaration in Logic.h and hide it in Logic.cpp.
- uses named enum instead of uint8_t to prevent misuse of enum values.
- reorder Logic members to make much sense.

Please have a look!

MCUdude, 2019-10-11:
I'm closing this issue now as #50 is merged. @SpenceKonde would you be interested in this library? Feel free to modify the examples to match the tinyAVR hardware better.

closed 2019-10-11 by MCUdude