MCUdude/MegaCoreX #48: Create an analog comparator library

MCUdude, 2019-10-10:
It would be nice if we had an efficient, easy to use library for the analog comparator (AC). It's way more advanced than on the older AVRs, and internal reference voltages can be connected, so external parts can in some cases be eliminated. Maybe it could be made compatible with tiny0/tiny1 too?

I will start out with something that's build the same way as the Logic library.

The new megaAVRs are pretty neat, let's make its awesome peripherals more available for the average Arduino user!

MCUdude, 2019-10-17:
I've just pushed come library files for interfacing with the analog comparator (eb5c703). No examples are ready yet, but for those who are curious, here's a simple demo:

``` c++
#include <Comparator.h>

void setup() 
{
  Comparator.output = out::enable;      // Enable output on digital pin 7 (PA7)
  Comparator.hysteresis = hyst::large;  // set +-50mV hysteresis
  Comparator.input_p = in_p::in0;       // Use positive input 0
  Comparator.input_n = in_n::dacref;    // Use the DACREF as the negative signal
  Comparator.reference = ref::vref_2v5; // Set the DACREF voltage reference to 2.5V
  Comparator.dacref = 0xff;             // Don't "divide" down the voltage

  Comparator.init();
  Comparator.start();
}

void loop() 
{

}
```

The tiny0/tiny1's have _very_ similar analog comparator hardware, but the tiny1's have three separate comparators instead of one for the mega0/tiny0. Because the input and output pins are all over the place on the tiny0/tiny1's (just like the CCL pins), I haven't implemented support for them yet? @tgtakaoka do you know how we can handle these pins in an elegant way?

MCUdude, 2019-10-25:
@tgtakaoka is this something you'd like to test work on?

tgtakaoka, 2019-10-26:
Sorry, I don't have analog testbed in hand and can't try out the library.

MCUdude, 2019-10-26:
That's fine :) If @SpenceKonde wants to add the library to megaTinyCore it should be a fairly straight forward job.

closed 2019-10-26 by MCUdude