MCUdude/MegaCoreX #49: TCB2 counter interrupt

JarrodC-jpg, 2019-10-10:
I have been using TCB2 in periodic interrupt mode and the timing is inconsistent. Is anything else in the MegaCoreX using a TCB2 counter?

MCUdude, 2019-10-10:
Tcb2 is used to generate millis and micros. What does your code looks like?

JarrodC-jpg, 2019-10-10:
Also I am not using the most current commit. I am using the commit from September 17 (commit: 8268451a7759c22ca4f43fe5eb99ed49ee321944
This is the timer set up code, ISR code, and clock manipulation code to run the 4809 at 20MHz

` //timer set up
 void timer_init() {
 TCB2.CTRLB = TCB_CNTMODE_INT_gc;    // Setting to periodic interrupt mode
 TCB2.INTCTRL = TCB_CAPT_bm;			// enable interrupt
 TCB2.CCMP = Cnts;				    // setting
 TCB2.CTRLA = TCB_CLKSEL_CLKDIV2_gc | TCB_ENABLE_bm;	// timer prescaler of 2, Enable timer TCB2
 sei();
 }

 //Interrupt Compare match
 ISR(TCB2_INT_vect) {
 volatile int tSREG = SREG;           // saving the status register
 TCB2_INTFLAGS = TCB_CAPT_bm;	// clear flag

 if (t_count >= delayTime && pulseOut < count) {
 PORTB.OUT ^= PIN5_bm;               //tap out pin
 PORTC.OUT ^= PIN3_bm;
 t_count = 0;
 pulseOut++;
 } else {
 t_count ++;
 }
 SREG = tSREG;
 }

 void clk_init(void)
 {
 CPU_CCP = CCP_IOREG_gc;	// write IOREG to CCP to unlock MCLKCTRLB for 4 clock cycles
 CLKCTRL_MCLKCTRLB = 0;	// set CLK_PER to 20MHz
 }`

JarrodC-jpg, 2019-10-10:
I am guessing I will need to use a different timer/interrupt. Which ones are not being used?

MCUdude, 2019-10-11:
You can use TCB0, TCB1 or TCB3. They are however set up to be used as PWM outputs, so it might be a good idea to clear the the associated registers before using the timer.

MCUdude, 2019-10-17:
Did you have any progress?

MCUdude, 2019-10-27:
It would be nice if you could post your _entire_ code and explain what you're trying to do with it. If you do, I'll load it up and see if I can find any obvious issues.

JarrodC-jpg, 2019-10-28:
Using TCB0 resolved my issue. My apologies for not closing the issue.

closed 2019-10-28 by JarrodC-jpg

MCUdude, 2019-10-28:
No worries, thanks for the update!