MCUdude/MegaCoreX #52: PWM TImers

JarrodC-jpg, 2019-10-28:
Hey not an issue just more of a question. (If there is a better place to ask questions without submitting an issue please let me know, I am fairly new to using github).

Which timer controls the pwm output of pin 39 on the 4809, 48 pin configuration.  I need to implement another timer but I'm using pin 39 as a pwm output and do not want change pin numbers as I already have a board printed.

Thank you for taking the time.

MCUdude, 2019-10-28:
> If there is a better place to ask questions without submitting an issue please let me know, I am fairly new to using github.

Hi! I'm afraid not. There's note MegaCoreX thread on the Arduino forum (yet), so this is the best place to ask questions at the moment. 

If you want to know what timer is used for what pin, you can look it up in the [pins_arduino.h file](https://github.com/MCUdude/MegaCoreX/blob/820181de51b6e86012ac764ae2fed002a698f2c3/megaavr/variants/48pin-standard/pins_arduino.h#L273-L375). To directly answer your question, PWM on D39/PF5 is generated by TCB1.

JarrodC-jpg, 2019-10-28:
Thank you so much!

closed 2019-10-28 by JarrodC-jpg