MCUdude/MegaCoreX #55: Path specification for avrdude is wrong

egilkv, 2019-11-17:
I believe it should be:

tools.avrdude.path={runtime.tools.avrdude.path}
tools.avrdude.cmd.path={runtime.tools.avrdude.path}/bin/avrdude

The {path} thing no longer works in my environment, and it does seem wrong from the outset. Not sure if the first line here is really needed either.

PS: The, excuse my french, rather stupid swap() functions that were introduced causes enough of git heartache for me - I am no git wizard - so I probably will no longer bother to generate pull requests.

MCUdude, 2019-11-17:
> tools.avrdude.path={runtime.tools.avrdude.path}
tools.avrdude.cmd.path={runtime.tools.avrdude.path}/bin/avrdude

> The {path} thing no longer works in my environment, and it does seem wrong from the outset. Not sure if the first line here is really needed either.

Sure! Could you please post the error you get?

> PS: The, pardon my French, rather stupid swap() functions that were introduced causes enough of git heartache for me - I am no git wizard - that I probably will no longer bother to generate pull requests.

Can't you just fork this repo again? I'm definitely no git wizard either. I personally line swap() better than pins(). But you're not willing to create new PRs because I added code you think are stupid?

uzi18, 2019-11-20:
@MCUdude you got it here: #56

closed 2019-11-21 by MCUdude