MCUdude/MegaCoreX #59: Serial baud rate seems incorrect on internal 20MHz

tgtakaoka, 2020-01-18:
As long as I tried with 
- the latest MegaCoreX on github repo
- ATmega4809-p (DIP40)
- Internal 20MHz
- 40/48 pin standard
- Optiboot (UART1 default pins)
- Examples > 04.Communication > ASCIITable
  (with modification with #define Serial Serial1)

Not only default 9,600bps of ASCIITable but also other bps, such as 115,200bps, produces garbage on Serial Monitor.

Other clocks, Internal 16MHz to 1MHz with any baud rate work fine.

I hope this is a bug and will be fixed soon.

Thanks,
Tadashi.

MCUdude, 2020-01-18:
Thanks for reporting! I'm sure there's a bug in there somewhere, but unfortunately, I don't have much time to investigate this at the moment. Feel free do dive in if you like!

MCUdude, 2020-04-19:
I tested this on actual hardware recently, and it worked just fine at 20 MHz. It may be that your particular chip has an oscillator that's a little out of tune.

tgtakaoka, 2020-04-20:
Hi, dude/, thank you for the confirmation.
I too confirmed that the latest tip-of-toe MegaCoreX + optiboot works fine with 20MHz internal clock of ATmega4809P. The issue was may be due to my serial host machine (Chromebook + crouton).

So let me close this as not-reproducible.

closed 2020-04-20 by tgtakaoka