MCUdude/MegaCoreX #60: SPI, signal Slave Select in INPUT?

rlivio, 2020-02-01:
Inside the file SPI.cpp, in SPIClass::begin(), the line SS is programmed as an input.

`pinMode(PIN_SPI_SS, INPUT);`
`pinMode(PIN_SPI_SS_PINSWAP_1, INPUT);`
`pinMode(PIN_SPI_SS_PINSWAP_2, INPUT);`

This don't work for me, I have to change in OUTPUT and now the SPI work correctly.

Livio

MCUdude, 2020-02-01:
Seems like this was fixed some months ago in the official repo:
https://github.com/arduino/ArduinoCore-megaavr/commit/e6410c9a79171386f5da98ef6630455ca4d3379b

Does it work for you if you remove this line completely?

rlivio, 2020-02-01:
Yes, it works if I remove those lines. 
In my case the SS line is programmed as output in MCP23S17.cpp, before call the begin() in SPI.cpp

MCUdude, 2020-02-01:
Great! I'll fix it immediately.

closed 2020-02-01 by MCUdude

rlivio, 2020-02-01:
I am not sure that removing those lines is the best solution, if the SPI work as master is better to program SS as output inside SPI.cpp where also MOSI and SCK are programmed as output.

MCUdude, 2020-02-01:
The thing with SPI is that the chip select (CS) can be any pin. SS is only used when the microcontroller acts as an SPI slave.

When in master mode the user has to handle chip select manually. Just like in this Arduino example::
https://github.com/MCUdude/MiniCore/blob/master/avr/libraries/SPI/examples/DigitalPotControl/DigitalPotControl.ino

The reason for this is because SPI is a bus that may have multiple devices attached to it. The only way to interact with a slave is to dedicate a chip select pin.

rlivio, 2020-02-01:
Ok, I never use as a Slave.
You're right, this is the reason I found the instruction to put SS in output inside MCP23S17.cpp.

commit 41accb44 (2020-02-01): Fix SPI SS pin issue
Closes #60