MCUdude/MegaCoreX #61: Mistake on MegaCoreX 40-pin minimal setup circuit

technoblogy, 2020-02-22:
The TXD2 and RXD2 lines should be swapped; TXD2 goes to RXD on the FTDI, and RXD2 goes to TXD on the FTDI.

By the way, while I'm here, why do old AVR circuits use a 10kΩ resistor from the DTR capacitor/reset to VCC, whereas your circuit uses a diode?

MCUdude, 2020-02-22:
> The TXD2 and RXD2 lines should be swapped; TXD2 goes to RXD on the FTDI, and RXD2 goes to TXD on the FTDI.

Yep, I see that now. It's the labels on the FTDI connector. They're supposed to be swapped.

> By the way, while I'm here, why do old AVR circuits use a 10kΩ resistor from the DTR capacitor/reset to VCC, whereas your circuit uses a diode?

There is supposed to be a 10k resistor there as well. I will add this too. The diode is needed on old and new AVRs when doing auto-reset using a USB to serial adapter. The diode prevents voltage doubling caused by the capacitor when the DTR pin toggles.

MCUdude, 2020-02-22:
> The TXD2 and RXD2 lines should be swapped; TXD2 goes to RXD on the FTDI, and RXD2 goes to TXD on the FTDI.
>>Yep, I see that now. It's the labels on the FTDI connector. They're supposed to be swapped.

Actually, no. I placed an FTDI connector on the board. The FTDI adapter itself has Rx on pin 2, and Tx on pin three. By doing what I've drawn in the schematic (assuming that the FTDI adapter itself connects to the 1x6 pin connector), it will work.

technoblogy, 2020-02-22:
Yes, the wiring is correct but the labels on the FTDI adapter are the wrong way round.

MCUdude, 2020-02-22:
> Yes, the wiring is correct but the labels on the FTDI adapter are the wrong way round.

No, I don't have an FTDI adapter in my schematic, I have the connector that the FTDI adapter connects to. Thus, the labels are correct in my opinion. But the pull-up resistor is missing, yes

technoblogy, 2020-02-22:
Ah, I see what you mean now. Anyway I meant to say: thank you very much for your work on this great repository - it's really useful!

closed 2020-02-22 by technoblogy