MCUdude/MegaCoreX #63: SPI.begin() configures SS as an input

SpenceKonde, 2020-04-11:
Why? 

We are setting SSD in SPI0.CTRLB, so the master mode does not use the SS pin... (thank god for that addition to the SPI module... the classic avr behavior was so awful) so why would we set it as an input and potentially scribble over the user's configuration when the pin has no particular relevance to the master with SSD set... and even if we weren't setting that, the user would probable want it set OUTPUT anyway - or at least INPUT_PULLUP - certainly not a floating input!

MCUdude, 2020-04-12:
I've been looking trough the SPI.cpp file, and I do find the line where we disable the slave functionality, line 122.

https://github.com/MCUdude/MegaCoreX/blob/dff211f54f6cb87755f16671955dab03e7174e35/megaavr/libraries/SPI/src/SPI.cpp#L122

However, the pin is never set to input or output? Could you point me to the line that you mean causes this issue?

SpenceKonde, 2020-04-14:
Hmm, sorry about this - it looks like you must have fixed this? I saw it when I was implementing pin swapping, after replacing the copy of SPI and Wire libraries in my core with yours - but it doesn't seem, to be in the code in your repo currently. I think when I started work on this, I may have been working with an older version of your core than I thought I was

closed 2020-04-14 by SpenceKonde