MCUdude/MegaCoreX #66: digitalPinHasPWM and EXTERNAL_NUM_INTERRUPTS macros wrong

SpenceKonde, 2020-04-21:
for 28 and 32 pin parts, digitalPinHasPWM is wrong - it doesn't account for TCA0 being in SPLIT_MODE.

For all parts, EXTERNAL_NUM_INTERUPTS is set to 47. This should be set to the number of digital pins (which is never greater than 41!). Even assuming no user code relies on this, it still makes WInterrupts allocate an excessively sized array to store user functions, even when said functions could never exist.

MCUdude, 2020-04-21:
> for 28 and 32 pin parts, digitalPinHasPWM is wrong - it doesn't account for TCA0 being in SPLIT_MODE.

Thanks, I will fix this.

> For all parts, EXTERNAL_NUM_INTERUPTS is set to 47. This should be set to the number of digital pins (which is never greater than 41!). Even assuming no user code relies on this, it still makes [...]

Actually, I'm pretty sure the interrupt number should be 48 instead. The default Arduino implementation assumes that all ports have 8 pins, and that all pins are available. 8*6 = 48.
I opened an issue about the interrupt implementation [here](https://github.com/arduino/ArduinoCore-megaavr/issues/56), but I never got much response.

SpenceKonde, 2020-04-21:
Oh FFS did they really do that.... /facepalm

Ugh. 

47 is correct in that case, as there's no PF7.

MCUdude, 2020-04-21:
> for 28 and 32 pin parts, digitalPinHasPWM is wrong - it doesn't account for TCA0 being in SPLIT_MODE.

Just looked at this, and can't find anything wrong really.

The 28-pin parts only have PC0-PC3 broken out, and gives us only four PWM pins.
The 32-pin parts have PC0-PC3 + PF4 and PF5. This gives us six PWM pins.

MCUdude, 2020-05-03:
@SpenceKonde can you confirm this?

SpenceKonde, 2020-05-03:
You have to select a port for the PWM output, see 
15.3.5 PORTMUX Control for TCA

I can't fathom why you'd pick a port that doesn't have all six pins available though?

MCUdude, 2020-05-03:
Yes, of course, I should probably use a different port on these chips. Which ports would you prefer to use on these parts, and for your ongoing project?

MCUdude, 2020-05-03:
Perhaps PORTD?

MCUdude, 2020-05-03:
OK, this will be the new timer routing for the 28 and 32 pin parts. @SpenceKonde do you agree that this is the most compatible one?

```c
// Timer pin mapping from pins_arduino.h
#define TCA0_PINS PORTMUX_TCA0_PORTD_gc // TCA0 output on PD[0:5]
#define TCB0_PINS 0x00                  // TCB0 output on PA2 instead of PF4
#define TCB1_PINS 0x00                  // TCB1 output on PA3 instead of PF5
#define TCB2_PINS 0x00                  // TCB2 output on PC0 instead of PB4
#define TCB3_PINS PORTMUX_TCB3_bm       // TCB3 output on PC1 instead of PB5
```

SpenceKonde, 2020-05-03:
I think PORTD makes the most sense for TCA0 yeah... looking at the TCBs....

MCUdude, 2020-05-03:
Thanks. the 32-pin pars do have PF4 and PF4, but I'd rather prefer to use PA2 and PA3 instead, to be consistent compared to the 28-pin parts.

SpenceKonde, 2020-05-03:
I would rather like to get the TCB PWM pins onto PORTF instead of PORTA - though it saddens me to think of a type B timer reduced to generating 8-bit PWM no matter what pin it's outputting on....

MCUdude, 2020-05-03:
> I would rather like to get the TCB PWM pins onto PORTF instead of PORTA

The problem is that the 28-pin parts only has PF0, PF1 and PF6. This means you can't route any timers to PORTF on this part.

I will assume people will migrate from a 28-pin DIP part to the 32-pin TQFP part. It would be great if these two packages were compatible, just like the ATmega328/P is.

closed 2020-05-05 by MCUdude