MCUdude/MegaCoreX #67: Cannot Install MegaCoreX

mohdrais, 2020-04-26:
Cannot install MegaCoreX either by URL or manual. I am using Arduino 1.8.12. By URL, the IDE print "Error downloading" whereas by manual nothing printed on the IDE. In the location  Program Files(x86)/Arduino/hardware there are folders 
1.arduino
2.tools
3.package_index_bundled.json
4.platform.keys.rewrite
By manually, I put the MegaCoreX in the /Documents/Arduino/hardware as being told.
TQVM

MCUdude, 2020-04-26:
Can you post a screenshot of the download error? There's no reason why it shouldn't show up in the IDE. Have you made sure that your sketchbook location is set to /Documents/Arduino?

MCUdude, 2020-04-27:
By looking at your other issues here, it seems like you were able to install though boards manager. Reopen if needed.

closed 2020-04-27 by MCUdude