MCUdude/MegaCoreX #68: Error Compiling for board ATmega4809

mohdrais, 2020-04-27:
Dear Sir, 
I tried to compile a sketch for Nano Every by using Arduino 1.18.12. Below is the error message:
Arduino: 1.8.12 (Windows 10), Board: "ATmega4809, Nano Every, GPIO, BOD 2.6V, Internal 16 MHz, No bootloader"

In file included from C:\Users\User\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.1\cores\coreX-corefiles/UART.h:28:0,

                 from C:\Users\User\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.1\cores\coreX-corefiles/Arduino.h:142,

                 from sketch\CorexPWM.ino.cpp:1:

C:\Users\User\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.1\variants\nano-every/pins_arduino.h:145:28: error: 'PIN_A15' was not declared in this scope

 static const uint8_t A15 = PIN_A15;

                            ^~~~~~~

C:\Users\User\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.1\variants\nano-every/pins_arduino.h:145:28: note: suggested alternative: 'PIN_A10'

 static const uint8_t A15 = PIN_A15;

                            ^~~~~~~

                            PIN_A10

exit status 1
Error compiling for board ATmega4809.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

ednieuw, 2020-04-27:
I have the same errors  in this ino file
https://github.com/ednieuw/DCFtiny-clock/tree/master/DCF_HC12TransmitterV38d

[Errors.txt](https://github.com/MCUdude/MegaCoreX/files/4539424/Errors.txt)

MCUdude, 2020-04-27:
There is a bug in the pins_arduino.h file for the Nano Every in the 1.0.1 release. a new release, 1.0.2 with this fixed will be available very soon, hopefully today.

ednieuw, 2020-04-27:
Hi Hans

Thanks a lot for repairing the bug so fast. 
Cheers
Ed

——————————————————————————————————-
Ed Nieuwenhuys |  Badhoevedorp, The Netherlands |  ednieuw@xs4all.nl   
https://ednieuw.home.xs4all.nl
——————————————————————————————————-

MCUdude, 2020-04-27:
Boards manager release 1.0.2 is now available. Please install it though boards manager and report back if it solved your issue 🙂

closed 2020-04-27 by MCUdude

ednieuw, 2020-04-27:
Checked. Errors are gone. Thanks