MCUdude/MegaCoreX #69: bool conversion error digitalWrite(DCF_LED_Pin, !digitalRead(DCF_PIN));

ednieuw, 2020-04-27:
There is a conversion error that runs well with the Arduino Nano every board

 digitalWrite(DCF_LED_Pin, !digitalRead(DCF_PIN));  
Error -->   cannot convert 'bool' to 'PinStatus' for argument '2' to 'void digitalWrite(pin_size_t, PinStatus)'

digitalWrite(DCF_LED_Pin, 1-digitalRead(DCF_PIN));    // --> Works

See also
https://github.com/ednieuw/DCFtiny-clock/tree/master/DCF_HC12TransmitterV38d

rocketscream, 2020-04-27:
Another workaround is if you use the CHANGE constant instead of reading the pin status.

MCUdude, 2020-04-27:
This will be fixed in the next release, v1.0.2, which will be available very soon. Meanwhile, you can use CHANGE instead.

ednieuw, 2020-04-27:
Thanks!

——————————————————————————————————-
Ed Nieuwenhuys |  Badhoevedorp, The Netherlands |  ednieuw@xs4all.nl   
https://ednieuw.home.xs4all.nl
——————————————————————————————————-

MCUdude, 2020-04-27:
Boards manager release 1.0.2 is now available. Please install this version and see if it fixes you issue

closed 2020-04-27 by MCUdude

ednieuw, 2020-04-27:
Checked. Errors are gone.