MCUdude/MegaCoreX #71: Interrupt on pin 2 is not working with DCF77

ednieuw, 2020-04-28:
I suspect the interrupt on pin 2 is not working

The program decodes the DCF77 signal into a time
For decoding it makes use of the DCF77 library of Thijs Ellenbaas and my own non-interrupt algorithm
When compiling with the Arduino Every the Th:counter is comparable in amount with the Ed: counter
Th: is the interrupt driven library and Ed: my non-interrupt receiver

Arduino Every 
@ 10:37:02 28-04-2020 DCFeff:89 S:L002
T103703 LDR:269=10% DCF Ed:33 Th:43  Both:32  Tot:47 10:37:07

MegaCoreX
@ 09:22:02 28-04-2020 DCFeff:75 S:L002
T092203 LDR:255=9% DCF Ed:463 Th:0  Both:0  Tot:614 09:22:07

I have tried DCF_INTERRUPT 2 and also tried DCF_INTERRUPT 0 because in your Nano Every pinout I see for pin2:   PA0 2 0IN0 TXD2 CLKI

The DCF77-interrupt is started as below:
-------
                    #ifdef ARDUINO_AVR_NANO_EVERY
#define DCF_INTERRUPT 2              // DCF Interrupt number associated with DCF_PIN ( 2 Nano Every)
                    #else if
#define DCF_INTERRUPT 0              // Nano Uno etc 
                    #endif      
DCF77 DCF = DCF77(DCF_PIN,DCF_INTERRUPT,LOW); 
------
See program: https://github.com/ednieuw/DCFtiny-clock/tree/master/DCF_HC12TransmitterV38d

MCUdude, 2020-04-28:
I know nothing about your application and the provided source code is just too much for me to go through.

Do your application work as expected when using the official Nano every board option?
If you're using Arduino pin 2, your interrupt number is 2.

ednieuw, 2020-04-28:
Yes, I use interrupt number 2.
Yes, the program works fine when using the official Nano every board compiler. 

This is, I think, the essential part of the DCF77 lib
```

DCF77::DCF77(int DCF77Pin, int DCFinterrupt, bool OnRisingFlank) 
{
	dCF77Pin     = DCF77Pin;
	dCFinterrupt = DCFinterrupt;	
	pulseStart   = OnRisingFlank ? HIGH : LOW;
	
	if (!initialized) {  
		pinMode(dCF77Pin, INPUT);	
		initialize();
	  }
	initialized = true;
}

/**
 * Start receiving DCF77 information
 */
void DCF77::Start(void) 
{
	attachInterrupt(dCFinterrupt, int0handler, CHANGE);
}
```

MCUdude, 2020-04-28:
Thanks. Is it only pin 2 that doesn't work, or is this an issue you see with all pins?

ednieuw, 2020-04-28:
I only use the interrupt of pin 2. 
All the other pins I use are with pinmode INPUT_PULLUP or  OUTPUT. They all work as expected with input and output.

Because I use pin2, DCF_PIN,  also with  pinMode(DCF_PIN, INPUT_PULLUP); in the setup I checked if this might caused the problem. But removing this line did not helped.

MCUdude, 2020-04-28:
I just tested this simple proof of concept sketch, and it works just fine with MegaCoreX. This proves that there is nothing wrong with the interrupt functionality itself.

```c++

void setup()
{
  pinMode(2, INPUT_PULLUP);
  Serial.begin(9600);

  attachInterrupt(2, myfunc, CHANGE);
}

void loop()
{  
}

void myfunc()
{
  Serial.println("Interrupt!");
}

```

ednieuw, 2020-04-28:
Then I will try to find the problem in my program or the DCF library
But there is a difference between the two compilers. Maybe a setting somewhere
Keep you informed

ednieuw, 2020-05-14:
Solved the problem why DCF77 library was not working. 
The DCF77 InternalClockSync example did not work with neither compilers. 
The DCF77 library example needs, with the Nano EVERY, pinMode(DCF_PIN,     INPUT_PULLUP); in the setup(); 

The MEGAcoreX compiler does not recognize  ARDUINO_AVR_NANO_EVERY as define

```
`                    #ifdef ARDUINO_AVR_NANO_EVERY
#define DCF_INTERRUPT 2                   // DCF Interrupt number associated with DCF_PIN ( 2 Nano Every)
                    #else if
#define DCF_INTERRUPT 0                   // Nano Uno etc 
                    #endif      
DCF77 DCF = DCF77(DCF_PIN,DCF_INTERRUPT,LOW); // HIGH (HKW) or LOW (Reichelt). Some DCF modules invert the signal
 `
```

MCUdude, 2020-05-14:
Great to hear you figured it out.

> The MEGAcoreX compiler does not recognize ARDUINO_AVR_NANO_EVERY as define

That's right. But the official Nano Every entry doesn't either. With MegaCoreX you can use `NANO_EVERY_PINOUT`:
https://github.com/MCUdude/MegaCoreX/blob/4e712716f00bb66337dee3efd58518fa7c788820/megaavr/variants/nano-every/pins_arduino.h#L29-L30

closed 2020-05-14 by MCUdude