MCUdude/MegaCoreX #73: Servo Library framerate bug

KHeintz, 2020-05-24:
Somewhere there is a bug in the Servo Library of MegaCorex. The framerate of 4ms is much too short and kills every analog servo. 
REFRESH_INTERVAL has no effect. If I use the original servo library the framerate is correct.
To use it you only have to add the timer TCB3 to the servo timer and to the servo.cpp.
Illustrated with an ATMega4809 40 Pin and the Knob example. 
Change to the Arduino servo library megaavr
Servo Timer.h add
#if defined(SERVO_USE_TIMERB3)
  &TCB3;
#endif
Servo.cpp add
#elif defined(SERVO_USE_TIMERB3)
ISR(TCB3_INT_vect)

[image]
[image]

SpenceKonde, 2020-05-30:
Posting on here so I get notifications - and to ask a few questions. 

This is really my problem to deal with, not @MCUdude's - his implementation of Servo is a direct copy of what I did for megaTinyCore in order to remove Servo's dependence on the prescaler of TCA0 (which is the sort of insane stupidity that makes you want to reach through the screen and slap the Arduino team for - right up there with making TCB-for-millis depend on the TCA0 prescaler too, when you can rig up TCB0 as millis so an overflow happens exactly once per ms, and then just rightshift the TCB count in micros - reconfiguring TCA0 to get custom speed PWM is not a particularly exotic thing to want to do...)

Can you explain a bit more what the constraints are here, w/regards to how often the signal is sent? What is the minimum refresh interval? What is the maximum? 

Also, what do you mean by "analog servo"? I did my testing with one of the ubiquitous MG90S servos from ebay. Do those tolerate some sorts of input that your "analog servos" do not?

KHeintz, 2020-05-30:
The problem is the refresh interval (frame time) does not work in this Servo Library.
The electronics of servos can be digital or analog. Accordingly, servos are classified as digital or analog servos. Digital servos have a higher angular resolution and the servo position is controlled faster and more precisely. The pulse width for both servo types is in the normal range between 1ms and 2ms.  A servo pulse of 1.5 ms width will typically set the servo to its "neutral" position (typically half of the specified full range), a pulse of 1 ms will set it to 0°, and a pulse of 2.00 ms to 90° (for a 90° servo). Frame times (refresh_interval) for analog servos are around ~20ms. 18-22ms are normal, all under 16ms are much to short. This is the "standard pulse servo mode" used by all analog servos. Longer frame times are not a problem. Much shorter frame times cause extreme power consumption and overheating. In addition, the servos can twitch uncontrollably or lose their neutral point or stop operation immediately.
Anlog
https://de.wikipedia.org/wiki/Servo#/media/Datei:TiemposServo.svg

KHeintz, 2020-05-30:
MG90S is a digital servo and the short frame time does not matter. Do the test with a HXT900 for example and soon it will be clear what I mean.

MCUdude, 2020-11-16:
Fixed in c2321a7

closed 2020-11-16 by MCUdude