MCUdude/MegaCoreX #75: Any chance of integrating MegaCoreX into PlatformIO?

tombull, 2020-06-06:
Your MiniCore, MightyCore and MegaCore are available as frameworks in PlatformIO's platform-atmelavr platform. Have you got any plans to integrate MegaCoreX into PlatformIO's platform-atmelmegaavr platform? I would love to see this. Is there anything I can do to help? Is it something I could do myself? Any pointers on how to get started?

MCUdude, 2020-06-06:
Cool surname! Exactly the same as mine, even though I'm Norwegian.

> Is there anything I can do to help?

I did put in a lot of effort to get Mega/Mini/Mighty/Major/MicroCore support working like *I wanted*, and I worked together with the PlatformIO developers to create a script that could calculate and set the correct fuses, and another script that would load the right bootloader. I am planning to do this for MegaCoreX as well, but it's a little more work I believe.

Here are some thoughts though:
* First we need the PlatformIO devs to add MegaCoreX. This is fairly straight forward, but they will have to do it
* Then the toolchain needs to be patched so that it uses the latest avr-libc, which fixes some compatibility issues, like #43. The latest staging Arduino toolchain has this patch applied.
* Now that we have basic MegaCoreX support, we could start working on the fuses/bootloader script. It should have the same parameters as the current script supports, and it should be built in a similar way so that it's easy to add new targets (ATtiny's, AVR-DA's). I would probably be the one that would have to do this.

These a few small issues I'd like to fix before doing a new release. The release, however, will be "Ready for PlatformIO".

tombull, 2020-06-06:
Thanks for the response - that sounds awesome!
I had a look through the `.py` files in platform-atmelavr and platform-atmelmegaavr and figured I could make the changes required - at least for my purposes... but I was struggling to figure out how I would add a framework package for MegaCoreX - it figures that only the PlatformIO guys can do that! If you want any help when the PlatformIO guys have done their bit, then let me know!

MCUdude, 2020-06-06:
> If you want any help when the PlatformIO guys have done their bit, then let me know!

Thanks, I'll let you know! I will probably open an issue soon over at the PlatformIO megaavr repo to see if any of the developers have some spare time for this

tombull, 2020-06-06:
There's already this issue... https://github.com/platformio/platform-atmelmegaavr/issues/2

MCUdude, 2020-11-04:
PlatformIO support is now present!

closed 2020-11-04 by MCUdude

tombull, 2020-12-01:
Finally got a chance to test this today on a basic project... it's amazing, thank you so much! 

In the time since I made the request, I have built my own build, library management system and CI system around arduino-cli with MegaCoreX... and it works OK, but not brilliantly. I'm really looking forward to transferring everything over to PlatformIO when I get some time.

Again, thanks so much for doing this work - it's going to make my programming life much better! And, of course, thank you for MegaCoreX - it's the basis of all of my projects for the ATmega4809, amazing work 👍 

Quick question (probably not the right place to ask, sorry): do you have any experience programming boards with anything other than jtag2updi? All my projects are getting up to the 48K limit of the ATmega4809, and programming is pretty slow (20-30 seconds to program and verify) so I'd love to be able to program the boards faster. Also, step-through debugging would be amazing. Just got an Atmel-ICE, but it doesn't work at all on my Mac without some really hacky workarounds and testing with avrdude on windows just gives errors. Thought you might have some experience with other programmers and might be able to point me in the right direction?

MCUdude, 2020-12-01:
> Again, thanks so much for doing this work - it's going to make my programming life much better! And, of course, thank you for MegaCoreX - it's the basis of all of my projects for the ATmega4809, amazing work 👍

You're welcome! FYI most of the work was to make a foundation so it's easier to add other "megaAVR" cores such as megaTinyCore and hopefully DxCore in the future.

> Quick question (probably not the right place to ask, sorry): do you have any experience programming boards with anything other than jtag2updi?

To be honest, I don't have much experience with jtag2updi, other than programming the Nano Every. My goto-programmer for <48kB UPDI chips is the [microUPDI programmer](https://www.tindie.com/products/MCUdude/microupdi-programmer/), which I  happen to sell on Tindie. It's running the latest version of the Atmel mEDBG firmware and works with Avrdude and Atmel Studio.
It's not super fast, but it's reliable and also has built-in UART communication for simple debugging. I think breakpoint-debugging works too, but I haven't tried it.

> All my projects are getting up to the 48K limit of the ATmega4809, and programming is pretty slow

Apparently, this is a limit or the old JTAG protocol jtag2updi is based on:

>> It's major limitation is speed; it can't go over 115200 Baud, because the protocol lacks definitions for higher speeds. It's actually inferior to the STK500v2 protocol in this respect, this older standard can run at any speed avrdude instructs it to.

You can probably increase the upload speed by using [pyupdi](https://github.com/mraardvark/pyupdi). It's [supported in PlatformIO](https://github.com/MCUdude/MegaCoreX/blob/master/PlatformIO.md#pyupdi)!