MCUdude/MegaCoreX #76: Support for AVR DA

sstaub, 2020-06-13:
It seems that the new AVR DA chips are very similar too the megaAVR-0. Maybe you can also support them. The AVR128DA48 Curiosity Nano Evaluation Kit is a very interesting board I will start to play with it next month.

MCUdude, 2020-06-13:
I will not add AVR-DA support to this core, but I can reveal that a dedicated AVR-DA core is under development, but I'm not directly involved with the development. It will probably be announced over at the Arduino forum (microcontroller section) when it's ready.

closed 2020-06-13 by MCUdude

sstaub, 2020-06-13:
Thank you for that information. The reason why I thought that it is useful to add to this core, is that most of the examples on the microchip website are working for megaAVR-0 are the same for AVR-DA.