MCUdude/MegaCoreX #79: fuses questions

mikrocoder, 2020-06-25:
Hello,

your board package is great. I have two questions.

Where did you get the values for the additional BOD Level Fuses settings? It's not in the manual.  :-)

Why do you set the bit 2 in fuse 5 (0x05) to 1?
0b 1100.1101
Any particular reason? Because the bit is actually reserved and should be 0. Fortunately, the 1 has no negative effects.

MCUdude, 2020-06-26:
The additional fuse settings were something that Atmel Studio revealed. I'll assume that some BOD levels didn't work like expected/had some bugs, so they just removed them from the documentation. At the moment I'm not sure if I'll remove the additional BOD options or not. Have you tested them on your hardware?

When it comes to that bit in fuse 5, I honestly don't remember what it's for. I'm currently on vacation, and the computer I brought doesn't have Atmel Studio 7 installed. However, this fuse can also be found in Atmel Studio. At least in the old installation, I have on my other computer. I'll report back when I've checked this.

> The name for your BOD level 4.2V should actually be 4.3V. :-)

Thanks! Will fix this soon!

mikrocoder, 2020-06-27:
Hello,

I discovered the additional BOD levels in an older manual. I tried out your settings with an ATmega4808. They all work. You can leave them in there for all I care. I don't know if you really need them. For Ub 3.3V you could set BOD 2.9V instead of BOD 2.6V. I don't know if this is relevant in practice. For Ub 5V only BOD 4.3V remains.

Bit 1,2,4 and 5 in fuse register 0x05 is empty, unused.

Otherwise, the controller package is madness. All the work you put into it.
I can only say thank you.  :-)

MCUdude, 2020-06-28:
Now I remember why bit two in fuse 0x05 was set to one. When I was working with the microUPDI programmer I found out that it had a safe mode that prevented you from inserting incorrect fuse values. I remember trying to modify the SYSCFG0 fuse (fuse 0x05) in a way the programmer wouldn't let me. Instead, the programmer wrote 0xCD instead.

I don't have a microUPDI/mEDBG programmer at hand, but I remember it being very picky. I'll push a fix when I'm back from vacation and have done some testing with various fuse settings. The last thing I want is that the microUPDI/mEDBG refuses to write the fuses in boards.txt, and instead writes something else (0xCD for instance), and the user is left with a fuse mismatch error.

What programmer did you use to test this?

mikrocoder, 2020-06-28:
Hello,

I tried this with a jtag2updi programmer (https://github.com/ElTangas/jtag2updi) on my ATmega4808. And on my ATmega4809 Curiosity Nano Board. Everything works.

Do this in peace after your vacation. I wish you a pleasant holiday.

commit db2cfe32 (2020-06-30): Don't set unused fuse
Closes #79
closed 2020-06-30 by MCUdude