MCUdude/MegaCoreX #80: Problems of understanding the Serial Pinout

mikrocoder, 2020-07-11:
Hi,

I'm testing your package regarding the 3 serials on an Arduino Nano Every and I don't understand your pinout and documentary for that. Serial1 and Serial2 only output zero terminators but no text. With the pinout I have special problems why Tx0/Rx0 are double labeled. Because there is no Tx0/Rx0 at PB0/PB1.

At the pin PA0/PA1 the controller is actually connected to Tx0/Rx0.
The pin PB0/PB1 is actually connected to Tx3/Rx3 on the side of the controller.
Tx1/Rx1 is not connected to the controller side of the board.

From Arduino's point of view the alternative pins for Tx3/Rx3 (PB4/PB5) are used. This is called Serial in the Arduino IDE and is used for USB. So far I can understand this.

Another question. If you use your package with the Nano Every and configure it as Nano Every, is it possible to use your Serial1 and Serial2? Or how to read and understand your pinout? According to your documentation, Nano Every must be selected as the pinout.

Thanks a lot.

```
void setup (void)
{
    Serial.begin(9600);         // USB and -->> Pin Tx D9, Rx D10 ?
    Serial.println("\nStart");
    Serial1.begin(9600);        // Pin Tx D1, Rx D0
    Serial1.println("\nStart");
    Serial2.begin(9600);        // Pin Tx D2, Rx D7
    Serial2.println("\nStart");
}

void loop (void)
{
    serialOutput(1000);
}

void serialOutput(const unsigned int interval)
{
    static unsigned long lastMillis = 0;
    unsigned long ms = millis();

    if (ms - lastMillis >= interval)
    {
        lastMillis = ms;
        Serial.println("I am Serial.0");
        Serial1.println("I am Serial.1");
        Serial2.println("I am Serial.2");
    }
}
```

MCUdude, 2020-07-25:
Sorry for being so slow to reply. For some reason, I haven't been notified.

It took me a while to understand what you meant, but by going through the Nano Every pin definition file I found the issue you're pointing out. It as to do with some copy/paste errors from the Uno Wifi pinout. I will push a fix soon. Thanks for reporting!

I will rearrange Serial2 a little:
Available pin combinations for the *Nano Every* pinout are:

| Peripheral | Default                            | Alternative                  |
|------------|------------------------------------|------------------------------|
| Serial     | swap(0)  **or**  pins(25,24)       | swap(1)  **or**  pins(9,10)  |
| Serial1    | swap(0)  **or**  pins(1,0)         | swap(1)  **or**  pins(34,35) |
| Serial2    | swap(0)  **or**  pins(2,7)         | swap(1)  **or**  pins(28,27) |

closed 2020-07-26 by MCUdude

mikrocoder, 2020-07-26:
Hello,

no do not close yet. Thanks for reading and understanding. For longer texts I use DeepL. But unfortunately you have once again thrown the pin numbers coloured. Is it possible that you mixed the Arduino Every numbering with the standard 48 IC pinout? Because a Nano Every definitely does not have a pin 34 and 35. Is there a reason why you do not want to use the Arduino numbering as suggested? It would be unique on the Every. I haven't checked Uno Wifi Rev.2 yet.  :-)  :-)

MCUdude, 2020-07-26:
If you look in the **Alternative pins** section in the README, you'll see an updated table:

Available pin combinations for the *Nano Every* pinout are:

| Peripheral | Default                                                          | Alternative                                   |
|------------|------------------------------------------------------------------|-----------------------------------------------|
| Serial     | swap(0)  **or**  pins(25,24) <br/>(connected to USB-serial chip) | swap(1)  **or**  pins(9,10)                   |
| Serial1    | swap(0)  **or**  pins(1,0)                                       | swap(1)  **or**  pins(34,35) (not broken out) |
| Serial2    | swap(0)  **or**  pins(2,7)                                       | swap(1)  **or**  pins(28,27) (not broken out) |
| Serial3    | swap(0)  **or**  pins(6,3)                                       | swap(1)  **or**  pins(37,38) (not broken out) |

Note that no matter how I re-orient the Arduino pins, swap for Serial1 will never be available, since the pins aren't broken out on the Nano Every.

The official Nano Every implementation does not even support pin swapping.

mikrocoder, 2020-07-27:
Hello,

oh, yes. I'm confused now myself. I'm sorry. Too many pinouts.   :-)

commit a16ef7ca (2020-07-26): Fix and improve Nano Every pinout
Fix an issue where USART0 were incorrectly routed. Closes #80.
Swap Serial2 (hardware USART0) so the default RX and TX pins are routed to pins that are broken put