MCUdude/MegaCoreX #81: Error: 'BitOrder' has not been declared, does not name a type Compiling st77** lib example.

wa2mze, 2020-07-21:
I tried compiling the Example 'graphicstest' for the Adafruit ST77** TFT LCD driver and got several errors concerning 'BitOrder' not being declared,  (BitOrder dataorder = SPI_BITORDER_MSBFIRST)
BitOrder _dataOrder;

I used the atmega4809, 48 pin package as board type.  Strangely enough, selecting the Arduino Nano Every as the board type DOES compile without error, and they both use the same CPU.  This might be an issue with the ADAfruit library, but I can't find the definition for 'BitOrder' in the code.

MCUdude, 2020-07-22:
I tried to install the Adafruit ST7735/ST7789 library from the library manager, and I'm not able to make it build for any AVR platform. Can you post your full error messsage?

MCUdude, 2020-07-22:
I found a reference to `SPI_BITORDER_MSBFIRST` in the Adafruit Bus IO library.

try adding
```
typedef enum _BitOrder {
  SPI_BITORDER_MSBFIRST = MSBFIRST,
  SPI_BITORDER_LSBFIRST = LSBFIRST,
} BitOrder;
```

at the very top of your sketch, before all the #includes

MCUdude, 2020-07-25:
This has been fixed in the Adafruit BusIO library.

https://github.com/adafruit/Adafruit_BusIO/pull/32

closed 2020-07-25 by MCUdude