MCUdude/MegaCoreX #84: Unable to compile when using non-16MHz speed on Adafruit NeoPixel library

rocketscream, 2020-08-12:
Using any of the MCU in the core, it will fail to compile whenever the clock speed is not 16MHz. At clock speed of 8MHz it gives the following error:

```
Arduino: 1.8.13 (Linux), Board: "ATmega3209, 48 pin standard, Reset, BOD 2.6V, Internal 8 MHz, No bootloader"

/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp: In member function 'void Adafruit_NeoPixel::show()':
/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp:272:18: error: no match for 'operator|' (operand types are 'PORT_t {aka PORT_struct}' and 'uint8_t {aka unsigned char}')
       hi = PORTD |  pinMask;
                  ^
/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp:273:18: error: no match for 'operator&' (operand types are 'PORT_t {aka PORT_struct}' and 'int')
       lo = PORTD & ~pinMask;
                  ^
/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp:379:18: error: no match for 'operator|' (operand types are 'PORT_t {aka PORT_struct}' and 'uint8_t {aka unsigned char}')
       hi = PORTB |  pinMask;
                  ^
/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp:380:18: error: no match for 'operator&' (operand types are 'PORT_t {aka PORT_struct}' and 'int')
       lo = PORTB & ~pinMask;
                  ^
/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp:470:18: error: no match for 'operator|' (operand types are 'PORT_t {aka PORT_struct}' and 'uint8_t {aka unsigned char}')
       hi = PORTC |  pinMask;
                  ^
/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp:471:18: error: no match for 'operator&' (operand types are 'PORT_t {aka PORT_struct}' and 'int')
       lo = PORTC & ~pinMask;
                  ^
/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp:560:18: error: no match for 'operator|' (operand types are 'PORT_t {aka PORT_struct}' and 'uint8_t {aka unsigned char}')
       hi = PORTF |  pinMask;
                  ^
/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp:561:18: error: no match for 'operator&' (operand types are 'PORT_t {aka PORT_struct}' and 'int')
       lo = PORTF & ~pinMask;
                  ^
Using library Adafruit_NeoPixel at version 1.5.0 in folder: /Arduino/libraries/Adafruit_NeoPixel 
exit status 1
Error compiling for board ATmega3209.
```
For other speed other than 8MHz, it will complain speed is not supported  which is understandable. The error was compiled using a the simple example on the Adafruit NeoPixel library. Was trying to port some old code from board based on ATMega328P 8MHz. If you think this an issue of how the `#ifdef` was done on the library, please let me know.

MCUdude, 2020-08-12:
I'm not familiar with the Adafruit Neopixel library, so can't say what it is without a deep dive. But I was able to reproduce the issue. I've opened an issue over at the library repo.

closed 2020-08-12 by MCUdude

rocketscream, 2020-08-13:
Thanks Hans!