MCUdude/MegaCoreX #85: missing SERIAL_PORT_MONITOR, SERIAL_PORT_HARDWARE, SERIAL_PORT_USBVIRTUAL

bxparks, 2020-08-14:
Hi,
The following symbols are defined in  `megaavr/variants/uno-wifi/pins_arduino.h` and `megaavr/variants/nano-every/pins_arduino.h`:

* `SERIAL_PORT_MONITOR`
* `SERIAL_PORT_HARDWARE`
* `SERIAL_PORT_USBVIRTUAL`

They are missing from various variants in this Core: `32pin-standard`, `40pin-standard`, `48pin-standard`, `28pin-standard`. I use the `SERIAL_PORT_MONITOR` symbol in my programs and libraries to grab the correct `Serial` object, without worrying about the specific Core/board. These symbols are defined on classic AVR, SAMD, ESP8266, TeensyDuino, and obviously on megaAVR (curiously, not on ESP32).

Any thoughts about adding them to the variants in this Core?

MCUdude, 2020-08-14:
Sure, that's no problem.

The reason why they aren't added is that this core is chip oriented and not board oriented. I can, of course, assign these three macros to `Serial`, but that would not work if the user is using a 4809 Curiosity Nano, where the USB to serial adapter is connected to Serial3. So even though your code would compile it wouldn't really work with "boards" that don't use `Serial` as their "debug serial port".

bxparks, 2020-08-14:
Oh I didn't fully understand the organization of this Core. I see that under the "Pinout" menu, there is the "Nano Every", and this will pickup the `nano-every` variant, which contains the `SERIAL_PORT_MONITOR` definition. I think this will be sufficient for my needs. So you don't have to do anything, unless you think it will be useful for other situations.

MCUdude, 2020-08-14:
> Oh I didn't fully understand the organization of this Core. I see that under the "Pinout" menu, there is the "Nano Every", and this will pickup the nano-every variant, which contains the [...]

I think it's actually better, in this case, to not add them, since the `XXpin-standard` pinouts don't really have a default serial port. That's the whole idea of this core; to bring as many features as possible, and let the user decide what's best for particular hardware.

Now the compilation will throw an error, and that's because the standard pinouts don't have a default serial port. It's up to the user!

If someone _has_ to use one of the standard pinouts it's only a matter of manually defining these macros before your library/libraries are included

closed 2020-08-16 by MCUdude

bxparks, 2020-08-16:
Sounds good to me. Thanks for taking the time to look at this issue.