MCUdude/MegaCoreX #88: Using an 4808 as motor controller

NilsMinor, 2020-09-24:
Hi MCU Dude,

first of all thanks for creating the MegaCoreX Library! I want to start using the 4808 for a new project as motor controller. I need to control the speed of a motor using PWM and an impulse counter as Feedback. The speed is between 40 turns/min and 50 turns/min. 

I just wanted to ask for advice and what would you think is the best way to do this kind of task using an 4808. Should I use timers or the digitalFastRead ? Or the internal logic elements?

Thanks,
Nils

MCUdude, 2020-09-24:
Hi!

> first of all thanks for creating the MegaCoreX Library! 

You're welcome! Great to hear you find it useful.

> I want to start using the 4808 for a new project as motor controller. I need to control the speed of a motor using PWM and an impulse counter as Feedback. The speed is between 40 turns/min and 50 turns/min.

What kind of signal do you read from the motor? is it a continuous pulse, or is it some kind of pattern, similar to encoders? If there is a pattern, you can use the internal logic (Have a look at the [Logic library](https://github.com/MCUdude/MegaCoreX/tree/master/megaavr/libraries/Logic)) to trigger an interrupt when a special condition is met. If there are pulses, how many do you think there are per second? 10's 1000's? 100000's?

> Should I use timers or the digitalFastRead? Or the internal logic elements?

digitalReadFast/digitalWriteFast are good candidates for closed-loop motor control digitalReadFast/digitalWrite fast compiles down to a single instruction, so it is as fast as it can get. The CPU however, has to actually perform something in order to read or write to an IO. But you could definitely use digitalReadFast/digitalWriteFast instead of "regular" digitalRead/digitalWrite. However, I don't know enough about the application to tell if a dedicated timer is necessary too.

The internal logic is pretty neat and very powerful, and I've written a user-friendly library that makes it easy to get started. Give it a try and see if it can suit your application!

NilsMinor, 2020-09-25:
Hi,

thanks for your response :) 
The motor is driving an encoder wheel (diameter of 100mm) that has 50 holes. This gives me 50 impulses per revolution. The Target Speed is 45 turns per minute this gives me (45 turns * 50  impulses per turn / 60s) = 37,5 Hz. 

What do you think, the am mount if impulses is pretty slow isn't it? I think I will build a closed loop controller with the digitalFastRead functions or do you have a better advice?

Thanks,
Nils

MCUdude, 2020-09-26:
Th's pretty slow, yes. I think digitalReadFast/digitalWriteFast would be sufficient! You could of course experiment with the internal logic, but you definitely don't have to for this application

NilsMinor, 2020-09-28:
Hi Hans,
thanks for your help, I will give it a try :)

closed 2020-09-28 by NilsMinor

NilsMinor, 2020-10-29:
Hi @MCUdude ,
I have started to use my custom made hardware. From your docs I found that my 4808 (32 pin) can use the follwoing pins as PWM outputs (2, 3, 12, 13, 14, 15, 16, 17) but when using analogWrite, nothing happens. I selected either PD2 or PD7 to be my PWM output but without success. 

From pins_ file I found that only outputs like PC0 or PC1 are PWM pins (digital_pin_to_timer) so I am a bit confused on what to use. Should I follow the readme docs for using PWM?
Thanks,
Nils

MCUdude, 2020-10-29:
Hi!

That's weird. I just tested this sketch on my ATmega4808 (32-pin) test board, and measure all eight pins with an oscilloscope to see if the waveform looked correct. There was a PWM signal on all pins. Try this sketch and see if you get PWM output from the eight pins

```c
int brightness = 0;    // how bright the LED is
int fadeAmount = 5;    // how many points to fade the LED by

// the setup routine runs once when you press reset:
void setup()
{
  // Declare all PWM pins to be outputs
  pinMode(2, OUTPUT);
  pinMode(3, OUTPUT);
  pinMode(12, OUTPUT);
  pinMode(13, OUTPUT);
  pinMode(14, OUTPUT);
  pinMode(15, OUTPUT);
  pinMode(16, OUTPUT);
  pinMode(17, OUTPUT);
}

// the loop routine runs over and over again forever:
void loop()
{
  analogWrite(2, brightness);
  analogWrite(3, brightness);
  analogWrite(12, brightness);
  analogWrite(13, brightness);
  analogWrite(14, brightness);
  analogWrite(15, brightness);
  analogWrite(16, brightness);
  analogWrite(17, brightness);

  // change the brightness for next time through the loop:
  brightness = brightness + fadeAmount;

  // reverse the direction of the fading at the ends of the fade:
  if (brightness <= 0 || brightness >= 255) {
    fadeAmount = -fadeAmount;
  }
  // wait for 30 milliseconds to see the dimming effect
  delay(30);
}
```

NilsMinor, 2020-10-29:
Hi @MCUdude,
ok updating to 1.0.5 solved my problem :) Thanks a lot!

MCUdude, 2020-10-29:
As you can see in the [changelog](https://github.com/MCUdude/MegaCoreX/releases), I've done some PWM related improvements for the 28 and 32 pin parts.

Pro tip: use PIN_* macros to refer to pins instead of Arduino pin numbers. For instance:

```
digitalWrite(PIN_PA0, HIGH);
delay(1000);
digitalWrite(PIN_PA0, LOW);
delay(1000);
```

PIN_PA0 is just a macro that's defined in pins_arduino.h file. They always represent the corresponding Arduino pin, so migrating code from one chip to another is much easier. Especially since all megaAVR-0 chips have the same peripherals on the same pins.

https://github.com/MCUdude/MegaCoreX/blob/be6981622723d18b338504ba99ef3fb6dd2e77bf/megaavr/variants/32pin-standard/pins_arduino.h#L25-L51

NilsMinor, 2020-10-29:
Hi,

thanks for your help and your library! I will do like you explained and will use your macros.

Thanks a lot :)

NilsMinor, 2020-11-25:
Hey @MCUdude ,
sorry to reopen this topic but I have some trouble with my application and I need some advice from an expert.

My task is to control the speed of a vinyl player that's why I need to build an DC motor controller that is pretty stable (otherwise it sounds bad). My current approach is to count the speed of pulses and calculate the frequency in order to react on the changes in order to fit the target frequency of 27.77 and 37.5 to get 33/45 Hz. I am using a PID lib to calculate the output depending on the input. The system is very lightweight and the DC motor is not very powerful so it is not quite easy to get it stable.

Do you have an advice to measure the frequency in a very accurate way? I get stable pulses (low active) and I am using the timer setup from my code above. I am using the 20MHz setting for the 4808. Any idea how to get more accurate frequency (I think it is pretty good but maybe you know a better approach) and any Idea how to increase the PID speed that I am calling in my main loop?

Any Idea or advice is highly appreciated. Sorry for the wired question just hoping to get a new idea for this problem.

Best regards,
Nils