MCUdude/MegaCoreX #89: fix do_spm() vector in optiboot.

WestfW, 2020-09-27:
I think using the optiboot vector to write flash from the app working.  It's a bit different from traditional AVRs, but not TOO much...

Optiboot/optiboot@7975070
Neat test program, too.

MCUdude, 2020-09-27:
Thanks!

I have a few comments though.

First, the formatting of optiboot_x.c is all over the place. Tabs and spaces compile, ugh! Could you fix this and use either tabs or spaces? I personally prefer spaces though

Second, wouldn't it be better if cli.h/cpp was a c++ class, so that we can pass the actual HardwareSerial object to a function, for instance, begin()? The object itself doesn't need to be defined in the user sketch, but can be hidden in the h/cpp file like done with the Wire library. if class-based, the syntax can be a little more elegant too IMO.

```cpp
CLI.begin(Serial2);
CLI.reset(); // Instead of cliReset()

uint8_t myLine = CLI.getLine(); // instead of cliGetline()
```

WestfW, 2020-09-27:
Well, the tabs/spaces thing is inherited.  Perhaps I should have untabbified everything when I "forked" optiboot_x, but I didn't, and I'm loathe to make JUST that change.  "Coding standard" at the last place I worked said "tabs are every 8 spaces, and indentation may be some combination of tabs and spaces.  All reasonable editors can deal with it; figure out how.  New code can use only spaces, or a combo, but do not gratuitously change indentation."

And yes, cli should **definitely** be a class (and probably a template, to set up the buffer.)  That's why it hasn't been published as a separate library: "awaiting cleanup."  It used to be C, and only got renamed to .cpp to need less fiddling when working with Arduino code.

closed 2020-10-03 by MCUdude