MCUdude/MegaCoreX #91: Best way to mesaure frequency using an Atmega4808

NilsMinor, 2020-10-30:
Hi,
I am looking for an advice for counting / measuring the frequency I get from an photo transistor.  I get a frequency that is about 50-60 Hz and I need to measure this. I started by measure the delta time in an ISR whenever the transistor signal is falling.  But it feels like there is a better way of measuring the frequency.

My frequency pulse pin is connected to PIN_PA7 on my 4808 circuit. Any advice is highly appreciated :)
Thanks,
Nils

NilsMinor, 2020-10-30:
Hi, 
just FYI I found an old forum post that described my goal here that looked like 
```
void initTCB(void)
{
// pinMode(3, INPUT);                       // Port F, Pin 5 = Arduino ~D3       
   pinMode(9, INPUT);                       // Port B, Pin 0 = Arduino ~D9       

   PORTMUX.TCBROUTEA |= PORTMUX_TCB1_bm;    // PORTMUX Control for TCB: Route TCB1 signal to PF5 aka D3 (default is PA3/SCL aka D19)

// ~~~

   TCB1.CTRLB = 0 << TCB_ASYNC_bp           // TCBn.CTRLB: Asynchronous Enable: disabled
           // | 1 << TCB_CCMPEN_bp          // TCBn.CTRLB: Pin Output Enable: enabled
           // | 0 << TCB_CCMPINIT_bp        // TCBn.CTRLB: Pin Initial State: disabled
           // | TCB_CNTMODE_PWM8_gc;        // TCBn.CTRLB: 8-bit PWM
              | TCB_CNTMODE_FRQ_gc;         // TCBn.CTRLB: Input Capture Frequency measurement

// ~~~

// TCB1.DBGCTRL = 0 << TCB_DBGRUN_bp;       // TCBn.DBGCTRL: Debug Run (disabled)

// ~~~

   TCB1.EVCTRL = 1 << TCB_CAPTEI_bp         // TCBn.EVCTRL: Event Input Enable (enabled)
               | 0 << TCB_EDGE_bp           // TCBn.EVCTRL: Event Edge: (positive edge)
               | 1 << TCB_FILTER_bp;        // TCBn.EVCTRL: Input Capture Noise Cancellation Filter (enabled)

// ~~~

// TCB1.CTRLA = TCB_CLKSEL_CLKDIV1_gc       // TCBn.CTRLA: CLK_PER (CLKDIV1 aka No-Prescaling)
// TCB1.CTRLA = TCB_CLKSEL_CLKDIV2_gc       // TCBn.CTRLA: CLK_PER/2 (CLKDIV2)
   TCB1.CTRLA = TCB_CLKSEL_CLKTCA_gc        // TCBn.CTRLA: Use Clock from TCA
               | 1 << TCB_ENABLE_bp         // TCBn.CTRLA: Enable bit position (enabled)
               | 0 << TCB_RUNSTDBY_bp       // TCBn.CTRLA: Run Standby bit position (disabled)
               | 0 << TCB_SYNCUPD_bp;       // TCBn.CTRLA: Synchronize Update bit position (disabled)

               

// ~~~

   TCB1.INTCTRL = 1 << TCB_CAPT_bp;         // TCBn.INTCTRL: Capture Interrupt Enable (enabled)

// ~~~

// WARNING!
// By using the EVSYS, any event source, such as an external clock signal on any I/O pin, may be used as a control logic input.
// When an event action controlled operation is used, the clock selection must be set to use an event channel as the counter input.

// EVSYS.CHANNEL3 = 0b01001100;             // event channel 3 connects to pin PD4
// EVSYS.USERTCB1 = 4;                      // connect the counter to event channel 3 (4-1)

// EVSYS.CHANNEL5 = 0b01001101;             // event channel 5 connects to pin PF5 (Port F, Pin 5 = Arduino ~D3) WARNING! Not all generators can be connected to all channels!
// EVSYS.USERTCB1 = 6;                      // connect the counter to event channel 5 (6-1)

// EVSYS.CHANNEL0 = 0b01001000;             // event channel 0 connects to pin PB0 (Port B, Pin 0 = Arduino ~D9) WARNING! Not all generators can be connected to all channels!
// EVSYS.USERTCB1 = 1;                      // connect the counter to event channel 0 (1-1)

   EVSYS.CHANNEL1 = EVSYS_GENERATOR_PORT0_PIN7_gc;             // event channel 1 connects to pin PB0 (Port B, Pin 0 = Arduino ~D9) WARNING! Not all generators can be connected to all channels!
   EVSYS.USERTCB1 = 2;                      // connect the counter to event channel 1 (2-1)

// ~~~

   TCB1.CTRLA |= TCB_ENABLE_bm;             // TCBn.CTRLA: bit0=Enable bit mask 
}
```

```
ISR(TCB1_INT_vect)
{
   static bool toggle;
   
   counter++;

   period =  TCB1.CCMP;

   digitalWrite( PIN_LED_R, toggle ? HIGH : LOW); // diagnostic only

   toggle = !toggle;
}
```

```
void getPeriodValueFromISR () {
    uint16_t tempVal;
    cli();    // disable all interrupts
    tempVal = period;
    sei();    // enable all interrupts
    frequency = 1000000 / ((tempVal + 8) / 16) * 2;
}
```
This works great for me, what does an expert mean to this solution ?

MCUdude, 2020-10-30:
Since the frequency is so low, I don't really see the point of doing all that when the calculations are so simple.

The signal that goes into PIN_PA7, is it a clean squarewave that's easy to trigger from? If yes, you could just set up a simple `attachInterrupt` for that pin, and use millis() or micros() to measure how often it's run.

If the signal has a slow slope, I'll suggest using a comparator. The 4808 does have several internal ones, but none of these are connected to PIN_PA7. If you moved the signal to another pin, you could use the [Comparator library](https://github.com/MCUdude/MegaCoreX/tree/master/megaavr/libraries/Comparator), set up an internal voltage as a reference, and use the comparator attachInterrupt.

Here's a quick example showing a signal connected to PIN_PD2. Every time the signal on PIN_PD2 goes high, it increases a counter by one. And the program prints out this value and clears the counter every second. Note that there are many ways to do this, but this will for at least 100kHz.

```c++
/***********************************************************************|
| This is the formula for the generated voltage:                        |
| Vdacref = (DACREF / 256) * Vref                                       |
|                                                                       |
| See Microchip's application note TB3211 for more information.         |
|***********************************************************************/

#include <Comparator.h>

uint32_t counter = 0;

void setup()
{
  // Configure serial port
  Serial2.begin(9600);

  // Configure relevant comparator parameters
  Comparator.input_p = in_p::in0;       // Use positive input 0 (PD2)
  Comparator.input_n = in_n::dacref;    // Connect the negative pin to the DACREF voltage
  Comparator.reference = ref::vref_2v5; // Set the DACREF voltage to 2.5V
  Comparator.dacref = 255;              // Gives us 2.5V -> (255 / 256) * 2.5V = 2.5V
  Comparator.hysteresis = hyst::large;  // Use a 50mV hysteresis
  Comparator.output = out::disable;     // Use interrupt trigger instead of output pin

  // Initialize comparator
  Comparator.init();

  // Set interrupt (supports RISING, FALLING and CHANGE)
  Comparator.attachInterrupt(interruptFunction, RISING);

  // Start comparator
  Comparator.start();
}

void loop()
{
  if(millis() % 1000 == 0) // Run once every second
  {
    cli();       // Disable interrupts
    Serial2.printf("Input signal frequency = %ld Hz\n", counter);
    counter = 0; // Clear frequency counter
    sei();       // Enable interrupts
    delay(1);    // Prevent this from being executed twice
  }
}

// This function runs when an interrupt occurs
void interruptFunction()
{
  counter++;
}
```

NilsMinor, 2020-10-30:
Thanks this will help me a lot ! As the photo transistor does not deliver perfect edges I will use the Comparator and an other pin.
Thanks for your help!

closed 2020-10-30 by NilsMinor