MCUdude/MegaCoreX #93: Small issue Curiosity pinout

sstaub, 2020-11-07:
There is a small issue in the Curiosity Nano pinout diagram, the on board LED is connected to Pin PF5 not PA7.

classiphil, 2021-04-21:
Funnily enough, when I try to read A0 (PD0) it reads PF4?

MCUdude, 2021-04-21:
> There is a small issue in the Curiosity Nano pinout diagram, the on board LED is connected to Pin PF5 not PA7.

The LED labels are originally there to indicate that that is the default LED pin that LED_BUILTIN refers to. But it is a bit misleading and should be removed.

> Funnily enough, when I try to read A0 (PD0) it reads PF4?

Sounds weird. Can you post your code?

classiphil, 2021-04-21:
It was my mistake. I typed Analogread(A0) instead of (0).

Interesting though why (A0) leads to PF4...
Maybe A0 is interpreted as 100 (Hex A and a 0) so it is a binary 4?!? :D

MCUdude, 2021-04-21:
That shouldn't be possible. Be I need to know if you're using the 40-pin or 48-pin pinout.

classiphil, 2021-04-21:
The 40 Pin.

commit 31bc8f27 (2022-03-16): Fix incorrect LED pin on Curiosity Nano and Xplained Pro pinouts

Closes #93
closed 2022-03-16 by MCUdude