MCUdude/MegaCoreX #94: Frequency set wrong

mikkel75, 2020-11-09:
Hi,

I see an issue using the MegaCoreX lib on Arduino IDE (linux).
I'm using default frequency of internal 16 MHz.

Initially I observed this issue when communication on UART0. I saw transmitted data was sent too fast.
To get data correctly @9600baud I had to set Serial.begin(7500)!

I also made a simple LED blink with 1s delay - that was also too fast.

Anybody have an idea what could cause this?

Thanks!
M

mikkel75, 2020-11-09:
I just noticed that "Internal 10MHz" seems to work - but other internals don't... Maybe a hint to find the bug...?

MCUdude, 2020-11-09:
Sounds like you're using the wrong internal oscillator. You'll have to set the settings in the Arduino IDE and click Burn bootloader to set the correct settings

rocketscream, 2020-11-09:
Have been using with 8 and 16 MHz internal without issues.

mikkel75, 2020-11-09:
@MCUdude Ahhh I have to set that seperately? Why is that not part of the .hex?

MCUdude, 2020-11-09:
> @MCUdude Ahhh I have to set that separately? Why is that not part of the .hex?

Well, you'll have to ask Microchip about that. That's how some of the hardware settings are implemented in the AVRs, as separate fuses you can't set in the user application. There are two internal oscillators to choose from, a 16 MHz and a 20 MHz one. The prescaler can, however, be set in the user application. That's why you would get accurate timing if you chose the 20, 10, or 5 MHz oscillator option.
You can read more about how the fuses work in [Chapter 5.8 in the megaAVR-0 family datasheet](http://ww1.microchip.com/downloads/en/DeviceDoc/megaAVR0-series-Family-Data-Sheet-DS40002015B.pdf).

closed 2020-11-09 by MCUdude

mikkel75, 2020-11-10:
@MCUdude Many thanks for your answers.

When I read the main page for this lib I find "..However, all fuses are set every time you upload our program. ." - is that a mistake or what am I missing?

MCUdude, 2020-11-10:
> When I read the main page for this lib I find "..However, all fuses are set every time you upload our program. ." - is that a mistake or what am I missing?

When I'm uploading (using a UPDI programmer) it also sets the fuses. Can you turn on the verbose upload setting on the Arduino IDE preferences page and paste the entire Avrdude command + output?

mikkel75, 2020-11-10:
I'm using PyUPDI (https://github.com/mraardvark/pyupdi) on a RPi (3208 is connected on HAT), so not programming from within the IDE. I'm programming the .hex file with pyupdi.
I know (now) that the hex only contains the flash code - but are there any way to read the fuses from the output files?

MCUdude, 2020-11-10:
Ah, that explains it. Fuses are not placed in the hex file (they can be placed in .elf files), but have to be specified separately. I don't know how you could read the fuses from an elf file, but I think you can use pyupdi to read the fuse settings directly from the chip.

For reference, here is the Avrdude programming command:

```
/Users/hans/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude -C/Users/hans/Documents/Arduino/hardware/MegaCoreX/megaavr/avrdude.conf -v -patmega3208 -cxplainedmini_updi -Pusb -Uflash:w:/var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_36081/Blink.ino.hex:i -Ufuse0:w:0x00:m -Ufuse1:w:0x54:m -Ufuse2:w:0x01:m -Ufuse4:w:0x00:m -Ufuse5:w:0xCD:m -Ufuse6:w:0x06:m -Ufuse7:w:0x00:m -Ufuse8:w:0x00:m -Ulock:w:0xC5:m 
```

mikkel75, 2020-11-10:
OK thanks.
I can confirm that after reading back the fuses and correcting the FREQSEL from 0x2 to 0x1 the serial baud rate is correct.
So sorry the problem was not with MegaCoreX.