MCUdude/MegaCoreX #96: Set Atmega4808 internal clock without bootloader

NilsMinor, 2020-11-25:
Hello,
i am just wondering if I am doing right with the 4808. I program the 4808 with visual code and your Library and it work great! (thanks for that)
When I change the Clock source, how will this be applied to the 4808? (see image below)
[image]

How can I read the internal clock source? Thanks for your help!
BR,
Nils

MCUdude, 2020-11-25:
>  I program the 4808 with visual code

Woah, I've never seen a configuration window like this in VSCode. This isn't PlatformIO, right? Maybe some other plugin?

The ATmega4808 (and really all chips supported by this core) has three available clock sources that are set by the fuse bits; Internal 20 MHz, internal 16 MHz, and external clock. To get other frequencies using the internal oscillator, [a prescaler is used, and is done in software](https://github.com/MCUdude/MegaCoreX/blob/2b34004d185db90c5f115b0ae4c06fda10a1d831/megaavr/cores/coreX-corefiles/wiring.c#L343-L377), before setup() is ran.

I don't quite understand what you want to achieve by "reading" the internal clock source. This is a set-and-forget thing.

Off-topic:
Since you're already familiar with VSCode, I can absolutely recommend PlatformIO. I've worked together with the developers to get as much functionality out of it as possible. It's a bit more fiddly to set up, but once you're done doing that it's lovely to use.

MCUdude, 2020-11-25:
I have to learn to read properly before I reply...

> When I change the Clock source, how will this be applied to the 4808?

If 16 MHz, 8, 4, 2 or 1 MHz is used, it will use the internal 16 MHz oscillator and divide it down accordingly.

If 20, 10 or 5 MHz is used, it will use the internal 20 MHz oscillator and divide it down if accordingly.

NilsMinor, 2020-11-25:
Hey @MCUdude,
thanks for your response :) Yeah I am using vs code but it is not really stable (I have to restart the IDE several times). I will give it a try on my next project :D

 Ah ok I understand I was not sure if just selecting the right frequency from vs code will do the job, I was thinking that I need to burn some fuses in order to get the correct frequency :)
Thanks for your help !
BR,
Nils

closed 2020-11-25 by NilsMinor

MCUdude, 2020-11-25:
> Ah ok I understand I was not sure if just selecting the right frequency from vs code will do the job, I was thinking that I need to burn some fuses in order to get the correct frequency :)

You will have to "burn" the fuses again if you're changing between a 16 MHz and a 20 MHz base clock.

> Thanks for your help !

You're welcome!

NilsMinor, 2020-11-25:
I never burned the fuses I used it like it was manufactured ... Ho can I do this? ^^
I am not using any bootloader so I thought I can just select it from vs code but it make sense that I need to burn it. Can I do this with the Arduino IDE? I thought the frequency of 20MHz that I selected is right, as I am using Serial communication and i looks fine to me. How can I check what frequency was burned before?
Thanks :)

MCUdude, 2020-11-25:
> I never burned the fuses I used it like it was manufactured ... Ho can I do this? ^^

I'll assume you're using a UPDI programmer then. I Arduino IDE, just select the settings you want under the tools menu and simply hit upload. The fuses are set automatically along with writing the compiled hex file. There is possible to read the fuses in application, but I'm not sure how that is done. And why wouldn't you trust the settings you've applied in Arduino IDE? What you can do is to turn on the clock output feature. By doing this the main clock will be outputted on pin PA7.

Try adding this to your code and measure PA7 with a frequency counter or oscilloscope.
``` c
_PROTECTED_WRITE(CLKCTRL_MCLKCTRLA, CLKCTRL_MCLKCTRLA  | CLKCTRL_CLKOUT_bm); // Turn on clock output on PA7
```

NilsMinor, 2020-11-26:
Thanks for your help @MCUdude,
I will try this :D