MCUdude/MegaCoreX #99: Feature request: Board choice and bootloader for 3.3V boards

Miraculix200, 2020-12-15:
Flashing boards running on 3.3V fails because a baudrate of 115200 is too high, as far as I know.

While I was able to flash the bootloader using pyupdi.py, I can't use the Arduino IDE to upload sketches to it. It does write and read, but reports an error at the end (can't tell the exact error right now because I somehow damaged the Arduino IDE and it won't start). Changing baudrate to 9600 in boards.txt (or was it platforms.txt) will make it output "not in sync" errors.

I suppose this happens because the PCB I designed runs on 3.3V (and it would fry the NRF24L01+ in the circuit if I ran it on 5V).

I've spent the day trying to compile a bootloader with a different baudrate on Linux, but failed. I'll probably get it working sooner or later, but other people may have the same issue in the future.

MCUdude, 2020-12-16:
> Flashing boards running on 3.3V fails because a baudrate of 115200 is too high, as far as I know.

I guess you're referring to bootloader uploads, not UPDI uploads. 3.3V shouldn't really be a problem. The reason why 115200 baud was selected is that the bootloader switches to its internal 3.33MHz (20 MHz/6) during the upload sequence, and 115200 baud is the highest "common" baudrate supported at this clock frequency. 

3.3V, 115200 baud and 3.33MHz shouldn't really be a problem. I just tested this on real hardware, and I didn't experience any issues.

Miraculix200, 2020-12-16:
Oh ok. Maybe the problem is something else. I'll have a closer look again later. 

Yes, I'm refering to uploads through t he RX/TX pins (with a 3.3V FTDI board).

Miraculix200, 2020-12-16:
So this does not look like a baud rate problem? Writing and reading looks successful, but verification fails:
https://gist.github.com/Miraculix200/5a39d2a871013808a646831324f8189f

MCUdude, 2020-12-16:
But you can confirm it works like it should on 5V?

Miraculix200, 2020-12-16:
I'll try it with another board later, without soldering the NRF24L01+ on it. In 1-2 hours I'll know more.

Miraculix200, 2020-12-16:
Ok, it has nothing to do with the voltage. I have the same problem with 5V. 

Do I seperately have to set some fuses for it to work? This is how I flashed the MegaCoreX bootloader: 
`python.exe pyupdi.py -d mega4808 -c COM3 -f Optiboot_mega0_UART0_DEF_115200_A7.hex  -v -b 115200`

Miraculix200, 2020-12-16:
`Device info: {'family': 'megaAVR', 'nvm': 'P:0', 'ocd': 'D:1', 'osc': '3', 'device_id': '1E9650', 'device_rev': '0.1'}
Fuse:Value
0:0x00
1:0x00
2:0x02
3:0xFF
4:0x00
5:0xE4
6:0x07
7:0x00
8:0x00
9:0xFF
10:0xC5`

Does that look right?

MCUdude, 2020-12-16:
You have been using Pyupdi all along? It looks like you haven't set the correct fuse bits. You need to set the BOOTEND fuse to use the bootloader. You can make Arduino IDE calculate the fuses, but I'm not familiar with the pyupdi syntax, so I'm not sure how you set the fuses

Miraculix200, 2020-12-16:
Thanks. That fixed the problem.

So I flashed the bootloader using
`python.exe pyupdi.py -d mega4808 -c COM3 -f Optiboot_mega0_UART0_DEF_115200_A7.hex -v -b 115200`

and then set the BOOTEND fuse bit like this
`python.exe pyupdi.py -fs 8:0x02 -c COM3 -d mega4808`

Now uploading sketches works.

Wasn't able to flash the bootloader using Arduino IDE. I'm using a 3.3V FTDI board to flash it. Circuit looks like the ASCII schematic here: https://github.com/mraardvark/pyupdi

MCUdude, 2020-12-16:
> Wasn't able to flash the bootloader using Arduino IDE

At the moment, there's no support for pyupdi in Arduino IDE, so you'll have to use a dedicated programmer for this. You can, however, use pyupdi with PlatformIO.

Just bear in mind that you only sat one fuse. If timing functions isn't accurate, try setting the rest of the fuses

closed 2020-12-16 by MCUdude

Miraculix200, 2020-12-16:
Thanks for the information.

Here can can see the schematic and PCB design files of my project: https://oshwlab.com/Miraculix200/minimegazero24_copy