MCUdude/MegaCoreX #103: analogReadResolution() for 8 bit and 10 bit seems swapped

chansheunglong, 2021-01-12:
My setup is PIN_PF3 connected to a voltage divider (10M and 2.2M) connected to VCC (3.3V), the pin should read 0.595V, which is 406/1024 in 10 bit mode.
 
While trying the analogReadResolution() function, I found out the following:
When stock (did not call analogReadResolution()) : analogRead() return 403/1024
When analogReadResolution(8) : analogRead() return 403/256
When analogReadResolution(10) : analogRead() return 100/1024

Seems the setting is swapped?

MCUdude, 2021-01-12:
That is correct! I fixed this bug a few weeks ago, but I haven't published a new release. I'll do that in a few weeks. Just need to finish the [event system library](https://github.com/SpenceKonde/DxCore/issues/52) I'm working on first 🙂 

Analog read resolution issue: https://github.com/MCUdude/MegaCoreX/issues/101

chansheunglong, 2021-01-12:
Oops sorry for the duplicated issue, thanks for the quick response!

closed 2021-01-12 by chansheunglong