MCUdude/MegaCoreX #104: avrdude: AVR Part "atmega4809{upload.workaround}" not found.

bigginsking, 2021-01-15:
I'm able to burn a bootloader to an ATMEGA4809 (40 pin DIP) using jtag2updi, but when I try to upload a sketch I get the subject error:
```
avrdude: AVR Part "atmega4809{upload.workaround}" not found.

Valid parts are:
<deleted>
  m4809    = ATmega4809
  m48p     = ATmega48P
```

the part comes up in the list

The programmer I'm using works great on a ATTINY3217 with the MegaTinyCore.

I'm really excited at the idea of using the 4809, I'm out of I/Os on the 3217...

Any ideas on how to get a sketch to load?

Thanks!

Bill

MCUdude, 2021-01-15:
Hi! Can you turn on verbose upload in the IDE settings and post the entire Avrdude output?

bigginsking, 2021-01-15:
Thanks for the reply! Sure:

```
avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf"

avrdude: AVR Part "atmega4809{upload.workaround}" not found.

Valid parts are:
  uc3a0512 = AT32UC3A0512       [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:14657]
  c128     = AT90CAN128         [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:3461]
  c32      = AT90CAN32          [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:3845]
  c64      = AT90CAN64          [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:3653]
  pwm2     = AT90PWM2           [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:9045]
  pwm216   = AT90PWM216         [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:9318]
  pwm2b    = AT90PWM2B          [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:9238]
  pwm3     = AT90PWM3           [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:9228]
  pwm316   = AT90PWM316         [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:9265]
  pwm3b    = AT90PWM3B          [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:9252]
  1200     = AT90S1200          [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:1922]
  2313     = AT90S2313          [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:2151]
  2333     = AT90S2333          [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:2264]
  2343     = AT90S2343          [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:2392]
  4414     = AT90S4414          [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:2038]
  4433     = AT90S4433          [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:2516]
  4434     = AT90S4434          [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:2640]
  8515     = AT90S8515          [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:2723]
  8535     = AT90S8535          [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:2837]
  usb1286  = AT90USB1286        [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:11965]
  usb1287  = AT90USB1287        [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:12156]
  usb162   = AT90USB162         [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:12169]
  usb646   = AT90USB646         [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:11761]
  usb647   = AT90USB647         [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:11952]
  usb82    = AT90USB82          [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:12356]
  m103     = ATmega103          [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:2958]
  m128     = ATmega128          [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:3281]
  m1280    = ATmega1280         [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:10070]
  m1281    = ATmega1281         [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:10260]
  m1284    = ATmega1284         [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:4657]
  m1284p   = ATmega1284P        [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:4851]
  m1284rfr2 = ATmega1284RFR2     [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:10695]
  m128rfa1 = ATmega128RFA1      [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:10480]
  m128rfr2 = ATmega128RFR2      [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:10583]
  m16      = ATmega16           [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:4038]
  m1608    = ATmega1608         [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:15848]
  m1609    = ATmega1609         [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:15872]
  m161     = ATmega161          [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:6183]
[... cut ...]
  x64d3    = ATxmega64D3        [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:13804]
  x64d4    = ATxmega64D4        [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:13814]
  x8e5     = ATxmega8E5         [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:14500]
  ucr2     = deprecated, use 'uc3a0512' [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:14673]

```

MCUdude, 2021-01-15:
Thank! But right before this there is a command too that would be interesting to see 😊

bigginsking, 2021-01-15:
Sorry, how about this:
```
Sketch uses 510 bytes (1%) of program storage space. Maximum is 49152 bytes.
Global variables use 4 bytes (0%) of dynamic memory, leaving 6140 bytes for local variables. Maximum is 6144 bytes.
C:\Users\Bill\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude -CC:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf -v -patmega4809{upload.workaround} -cjtag2updi -PCOM5 -Ufuse2:w:0x02:m -Ufuse6:w:0x06:m -Ufuse8:w:0x00:m -Uflash:w:C:\Users\Bill\AppData\Local\Temp\arduino_build_552044/4809.ino.hex:i 

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf"

avrdude: AVR Part "atmega4809{upload.workaround}" not found.

Valid parts are:
  uc3a0512 = AT32UC3A0512       [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:14657]
  c128     = AT90CAN128         [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:3461]
  c32      = AT90CAN32          [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:3845]
  c64      = AT90CAN64          [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:3653]
  pwm2     = AT90PWM2           [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:9045]
  pwm216   = AT90PWM216         [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:9318]
  pwm2b    = AT90PWM2B          [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:9238]
  pwm3     = AT90PWM3           [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:9228]
  pwm316   = AT90PWM316         [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:9265]
  pwm3b    = AT90PWM3B          [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:9252]
  1200     = AT90S1200          [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:1922]
  2313     = AT90S2313          [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:2151]
  2333     = AT90S2333          [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:2264]
  2343     = AT90S2343          [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:2392]
  4414     = AT90S4414          [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:2038]
  4433     = AT90S4433          [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:2516]
  4434     = AT90S4434          [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:2640]
  8515     = AT90S8515          [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:2723]
  8535     = AT90S8535          [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:2837]
  usb1286  = AT90USB1286        [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:11965]
  usb1287  = AT90USB1287        [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:12156]
  usb162   = AT90USB162         [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:12169]
  usb646   = AT90USB646         [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:11761]
  usb647   = AT90USB647         [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:11952]
  usb82    = AT90USB82          [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:12356]
  m103     = ATmega103          [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:2958]
  m128     = ATmega128          [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:3281]
  m1280    = ATmega1280         [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:10070]
  m1281    = ATmega1281         [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:10260]
  m1284    = ATmega1284         [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:4657]
  m1284p   = ATmega1284P        [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:4851]
  m1284rfr2 = ATmega1284RFR2     [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:10695]
  m128rfa1 = ATmega128RFA1      [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:10480]
  m128rfr2 = ATmega128RFR2      [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:10583]
[... cut ...]
  x64d3    = ATxmega64D3        [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:13804]
  x64d4    = ATxmega64D4        [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:13814]
  x8e5     = ATxmega8E5         [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:14500]
  ucr2     = deprecated, use 'uc3a0512' [C:\Users\Bill\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.6/avrdude.conf:14673]

```

MCUdude, 2021-01-15:
For some reason, it seems like `{upload.workaround}` gets added to the target name. That's really weird, and I've never seen this before. MegaCoreX does not have any `upload.workaround`, but I know megaTinyCore does!

It gets added here: [megaTinyCore platform.txt#l128](https://github.com/SpenceKonde/megaTinyCore/blob/fffd432b0ee3b3d3d4673c3494677290a57895cc/megaavr/platform.txt#L128)

So for some reason, your install uses the megaTinyCore platform.txt  rather than the MegaCoreX one. @per1234 have you seen this before?

What happens if you uninstall megaTinyCore?

bigginsking, 2021-01-15:
Ah, the plot thickens. This is perhaps why I wasn't able to upload sketches to a Ardo Wifi Rev 2, perhaps the megaTinyCore is conflicting with the 4809? I don't know exactly what is happening, I'm pretty novice at this stuff but to program the Wifi Rev 2 I had to use another computer, I just couldn't get it to work on the machine that I was doing the 3217/megaTinyCore work on.

I'll try with the megaTinyCore uninstalled later today and report back.

bigginsking, 2021-01-16:
Ok, so I can upload when I remove the MegaTinyCore, and select the all upper case JTAG2UPDI programmer. I reinstalled the megaTinyCore and it made the all lower case jtag2updi programmer available and now it appears I can program both the 3217 and 4809 on the same computer using the same physical programmer, just swiching board and programmer in the IDE.

I'm good to go, and will close this out. Thanks for the help!

MCUdude, 2021-01-16:
Great to hear you figured it out! If you install the latest 1.8.13 IDE, it will only show compatible programmers.

bigginsking, 2021-01-16:
I'm really excited to use the 4809, I'm drawing up a board now, we'll see how it goes. This is just in time as it looks like supplies of the ATTINY3217 are pretty low for some reason. Thanks for making the core!

closed 2021-01-16 by bigginsking