MCUdude/MegaCoreX #105: Problem with Streaming Lib

mikrocoder, 2021-01-17:
Hello,

there is a problem with your package with the streaming lib.
https://github.com/janelia-arduino/Streaming
Redefinition errors are issued.
`error: redefinition of 'uint8_t get_value_width`

With the Standard Arduino IDE for the Nano Every there are no problems.
Is the problem really with your package or with the streaming lib?

MCUdude, 2021-01-17:
There is most likely an issue with the Streaming library. This core doesn't touch the uint8_t typedef.

A quick search for `get_value_width` in the Streaming library source reveals that it depends on `#define ARDUINO`. Try defining this before including Streaming.h

mikrocoder, 2021-01-17:
Thanks, I'll see what I can do.

mikrocoder, 2021-01-17:
On reconsideration, I again come to the conclusion that the problem is probably with your package. I compared the streaming lib with the default IDE setting for the Nano Every.  It uses ARDUINO_ARCH_MEGAAVR as an option like yours.
It works with it.
With your package I get said redefinitions errors. I think your package should be compatible to the standard and not vice versa. Or not?

MCUdude, 2021-01-17:
Ah, it's because of the custom F() macro implementation. I've reverted that now. You can either do a manual install of this core or wait until MegaCoreX v1.0.7 is released. It would probably only be a few days before the new release is ready.

closed 2021-01-17 by MCUdude

mikrocoder, 2021-01-18:
Why does everyone always close a request? It's not even finished yet. I did the update manual and get with the streaming test sketch
`StreamingExample.ino:1:
Streaming.h: In instantiation of 'void __FMT<Ft, T, Ts ...>::tstreamf(Print&, Ft) const [with Ft = const char*; T = __WIDTH<int>; Ts = {__WIDTH<int>, int, __WIDTH<int>, __WIDTH<int>, __WIDTH<int>}]':
Streaming.h:391:18:   required from 'Print& operator<<(Print&, const __FMT<Ft, T, Ts ...>&) [with Ft = const char*; T = __WIDTH<int>; Ts = {__WIDTH<int>, int, __WIDTH<int>, __WIDTH<int>, __WIDTH<int>}]'
StreamingExample.ino:37:59:   required from here
Streaming.h:359:41: error: invalid initialization of reference of type 'const __FlashStringHelper*&' from expression of type 'const char*'
  359 |     while(char c = get_next_format_char(format))
      |                                         ^~~~~~
Streaming.h:315:62: note: in passing argument 1 of 'char get_next_format_char(const __FlashStringHelper*&)'
  315 | inline char get_next_format_char(const __FlashStringHelper*& format_string)
      |                                  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~
`

mikrocoder, 2021-01-18:
Now I have restored the old state and do not understand the world. StreamingExample.ino compiles without errors. StreamingTests.ino generates a new error.  Are you testing both examples?

`StreamingTests:15:17: error: 'int8_t _Assert::compare(const char*) const' cannot be overloaded with 'int8_t _Assert::compare(const char*) const'
   15 |     int8_t      compare(const __FlashStringHelper* expected) const
      |                 ^~~~~~~
StreamingTests.ino:9:17: note: previous declaration 'int8_t _Assert::compare(const char*) const'
    9 |     int8_t      compare(const char * expected) const
      |                 ^~~~~~~
Bibliothek Streaming in Version 6.0.8 im Ordner: C:\Arduino IDE Portable\megaAVR0\arduino-1.8.13\portable\sketchbook\libraries\Streaming  wird verwendet
exit status 1
'int8_t _Assert::compare(const char*) const' cannot be overloaded with 'int8_t _Assert::compare(const char*) const'
`

MCUdude, 2021-01-18:
I closed the issue because I believe I fixed the problem. Both Streaming examples compile without any errors or warnings on my computer when compiling for ATmeg4809. I'm also using Streaming v6.0.8. Are you sure you didn't mess up the manual MegaCoreX install?

mikrocoder, 2021-01-18:
Hi,

You can wait a few days before closing for any further inquiry. If I had deleted the email with the link, I could not answer or would be forced to open a new thread. Both would be stupid for me and for you. Strangely enough, many people do this on Github. I can understand that "you" want to have your peace, but how to communicate then?

Please do not take it personally.

To the actual topic.  :-)

> Are you sure you didn't mess up the manual MegaCoreX install?
Good question. I have the content of the folder: 
`C:\Arduino IDE Portable\megaAVR0\arduino-1.8.13\portable\packages\MegaCoreX\hardware\megaavr\1.0.6`
and then deleted it.
After that I unzipped your **MegaCoreX-master.zip** and copied the content from the folder **megaavr**
into the folder:
`C:\Arduino IDE Portable\megaAVR0\arduino-1.8.13\portable\packages\MegaCoreX\hardware\megaavr\1.0.6`
copied. After that open the IDE fresh.

After restoring from the saved folder content at least everything works as before.

PS: Are you a native german speaker? If so, then we could speak german.  :-)

MCUdude, 2021-01-18:
> You can wait a few days before closing for any further inquiry. If I had deleted the email with the link, I could not answer or would be forced to open a new thread. Both would be stupid for me and [...]

Sorry for closing it so quickly. I usually don't do this (just look through the history if you're in doubt), but I'm usually quick to close the issue if someone has a trivial question or it's a fairly simple bug fix like this one was. I get the exact same notifications if an issue is closed or not, so the only difference is that the issue is closed, and we both can still comment on a closed issue.

I don't take it personally, I do realize I was a little quick, sorry!

> I have one additional question. I have already posted this on Arduino Github, but it may be completely lost.
https://github.com/arduino/ArduinoCore-avr/issues/386
Do you know why PA2/PA3 and PF2/PF3 are connected together? When is which port used?

These pins are physically connected together on the Nano Every for pin compatibility with the original Arduino Nano. By default, pin [PF2/PF3 is used as IO pins](https://github.com/MCUdude/MegaCoreX/blob/a4605d92f91eae17b0e9b74d2c7dd9715cc3abc6/megaavr/variants/nano-every/pins_arduino.h#L253-L254), and PA2/PA3 is used for i2c communication.

> PS: Are you a native german speaker? If so, then we could speak german. :-)

I'm a native Norwegian speaker, so English is preferred 😉

mikrocoder, 2021-01-18:
Thank you very much for your quick help. I prefer to wait for the automatic IDE update for your package.

Because I am working on my own Pin and TCB library. Therefore the questions.

With the thread closed, we now have a "secret" channel.   😉

mikrocoder, 2021-01-22:
Hello,

have made today the package update to 1.0.7. But unfortunately still get error messages with the streaming lib.
Can you please have a look at it?

[StreamingTest.txt](https://github.com/MCUdude/MegaCoreX/files/5859153/StreamingTest.txt)
[StreamingExample.txt](https://github.com/MCUdude/MegaCoreX/files/5859154/StreamingExample.txt)

MCUdude, 2021-01-23:
Well, I'm not able to reproduce the error. I've even removed my manual install and installed v1.0.7 through boards manager and uploaded it to a Nano Every. The streaming library works fine here and does what it should.

What happens if you add

```c++
#undef F
#define F(string_literal) (reinterpret_cast<const __FlashStringHelper *>(PSTR(string_literal)))
```

before `#include <Streaming.h>` ?

mikrocoder, 2021-01-23:
Hello Hans,

unfortunately the code insertion did not bring any improvement either.    ¯\_(ツ)_/¯
I have then completely set up the portable IDE again. Now everything works.   :-)
No idea what went wrong. Anyway, the main thing is that it works again.

Greetings to Norway from Germany.
Thanks for your patience.

commit ec43b446 (2021-01-17): Re-implement old F macro implementation
for compatibility reasons. Closes #105